NatLabRockies/ditto #51: Safety checks for models attributes

NicolasGensollen, 2018-04-19:
It is currently possible to set models attributes with incorrect types.
For example, ```Line``` objects have an attribute ```is_switch``` defined as an Integer:

```python
#From line.py in ditto.models
is_switch = Int(help='''This flag indicates whether or not the line is also a switch''', default_value=None)
```

Nonetheless, the following code does not raise any error or warning:

```python
model = Store()
line_obj = Line(model)
line_obj.is_switch = True
```

kdheepak, 2018-04-19:
If you `print(line_obj.is_switch)`, what do you get? If you get an integer, then `line_obj.is_switch` is coercing boolean to integer. I think this might be a issue specific to boolean and integer values. This might even be specific to a version of Python.

NicolasGensollen, 2018-04-19:
```python

In [1]: from ditto.models.line import Line

In [2]: from ditto.store import Store

In [3]: m=Store()

In [4]: line_obj = Line(m, name='test', is_switch=False)

In [5]: line_obj.is_switch
Out[5]: False
```

kdheepak, 2018-04-20:
And can you check what version of Python you are using?

NicolasGensollen, 2018-04-20:
The previous example was obtained using Python 2.7.13 and I get the exact same thing using Python 3.6.4.

NicolasGensollen, 2018-04-25:
Hey @kdheepak 
Another example of weird things with the ```Traitlets```:

If you do the following:

```python
#Import store and transformer objects
from ditto.store import Store
from ditto.models.powertransformer import PowerTransformer

#Create a store object
m = Store()

#Create an empty Transformer object
pf = PowerTransformer(m)

#phase_shift is defined as a DiTTo Float
pf.phase_shift = 2.3 #This does NOT raise an error

#reactances is defined as a List[Instance(Float)]
pf.reactances = [2.3] #raises an error
```
You get the following error:

```
TraitError: Element of the 'reactances' trait of a PowerTransformer instance must be a Float or None, but a value of 2.3 <class 'float'> was specified.
```

Now, I'm sure to understand why you can set a ```Float``` attribute with a Python ```float``` but not a list a ```Floats``` in the same way.

It looks like the following code does not raise an error:

```python
from ditto.models.base import Float

r = Float(m, 2.3)

pf.reactances = [r]
```

Also, the following code, which does something that should not be possible, does not raise any error:

```python
pf = PowerTransformer(m)

#Set the reactances as an empty list
pf.reactances = []

#Append non-float objects to the list 
pf.reactances.append("this is a string")
```

kdheepak, 2018-04-25:
The most recent concern in the latest PR.

[image]

That no longer throws an error.

Sidenote:

You should never use `.append` on the list of a traitlet trait, because it'll have an effect ONLY on that list. However, the data that that list represents can be stored on other traits on other objects and this change would not propagate. 

If I have to append something, I'd do the following.

```python
pf.reactances = [2.3]
r = pf.reactances
r.append(4.6)
pf.reactances = r
```

Does that make sense?

Also, you cannot use Float the way you used it above.

```python
from ditto.models.base import Float

r = Float(m, 2.3)

pf.reactances = [r]
```

This above code does not do what I think you intend to have happen. Don't initialize `Int`, `Float`, `List` or any other class from the `ditto.models.base` anywhere outside a class body/namespace. It will not work, because `Int`, `Float`, `List` etc are all traitlet [Descriptors](https://docs.python.org/3/howto/descriptor.html) that only work in a class body, and only when the class in question inherits from `traitlets.HasTraits`.

kdheepak, 2018-04-25:
Also, 

```python
>>> issubclass(bool, int)
 True
```

because the above is true, the first example will never throw an error. Traitlets just checks for the instance of the object that is passed in.

```
>>> isinstance(True, int)
 True

>>> isinstance(False, int)
 True
```

This is a Python "feature" unfortunately.

kdheepak, 2018-04-25:
See [PEP-0285](https://www.python.org/dev/peps/pep-0285/) for historical context. Python is not ideal for these types of type checks. Feel free to close this issue unless you find something else that isn't working as expected.

NicolasGensollen, 2018-04-25:
Thanks for addressing this issue so quickly and for all the explanations @kdheepak !
I think we can close this for now, although I'm still a little concerned with the ```append``` issue. I understand the problem, but I think that 99.9% of the users will write 

```python
pf.reactances.append(4.6)
```

instead of what you wrote because it feels more natural. 
This is probably not top priority for now though.

closed 2018-04-25 by NicolasGensollen

kdheepak, 2018-04-25:
We shouldn't use a `List` and should opt to use `Tuple` instead.