NatLabRockies/ditto #65: Name convention to avoid conflicts

NicolasGensollen, 2018-04-26:
Most tools like OpenDSS requires having unique names per object class. It means that you can have a transformer named ```Reg1``` and a regulator with the same name.

IEEE 13 node feeder example:

```
New Transformer.Reg1 phases=1 XHL=0.01 kVAs=[1666 1666] Buses=[650.1 RG60.1] kVs=[2.4  2.4] %LoadLoss=0.01
New regcontrol.Reg1  transformer=Reg1 winding=2  vreg=122  band=2  ptratio=20 ctprim=700  R=3   X=9
```

However, this raises an issue in DiTTo since names have to be unique.

We had these naming issues before, and the way we handled them was to add a prefix (like ```load_```) to the objects when we read them in. 

```python
#From OpenDSS reader
load_name = 'load_' + name.split('oad.')[1].lower()
```

Currently, the previous code creates a ```PowerTransformer``` object named ```Reg1``` and a ```Regulator``` object named ```regulator_Reg1```.

This is done in some readers and for some objects only. In other words, it is currently very inconsistent and hard to maintain... (same problem with upper/lower case)

I kind of like the idea of adding prefix to the names but there might be other ways to handle this. Any thoughts?

If we keep the prefix idea, I think we should add them consistently, for all objects, and remove them when writing out such that the names before and after match. 
We should also probably implement some mechanism to check that the prefix are consistent and valid.
Also, for maintainability, we should have a mapping ```<DiTTo object>: prefix``` stored somewhere and used by all parsers.

Thoughts?

NicolasGensollen, 2018-06-01:
Hey @kdheepak, we faced this issue again today with @tarekelgindy when looking at OpenDSS files exported from WindMill. Basically, every line has the same name as its from node. This is not causing any issue in OpenDSS but it is very problematic when reading into DiTTo since we need to have name uniqueness. I believe Aadil will also have the same problem very soon with the WindMill parsers.

I was wondering if you'd have an opinion on how to handle this?

kdheepak, 2018-06-01:
That is annoying. You can store `Transformer.Reg1` and `regcontrol.Reg1` as the names, but then you will have to split on `.` and discard the first part when using it as an attribute anywhere else. We cannot use a unique ID for every element since we don't know where it will be used later on in the circuit definition.

tarekelgindy, 2018-06-01:
One approach I was thinking about for this was to modify the decorator functions that use name lookups in store.py to also consider the type of an object when a lookup is made. This would require changing the underlying dictionary structure to index objects by both type and name. I'm not sure how well this would integrate into the codebase (e.g. if a reader/writer is looking up an object by name) but I suspect this probably more of an issue in the analysis or modification modules which might be easier to fix.

@kdheepak  what do you think about this?

NicolasGensollen, 2018-06-05:
I think this is more or less the same idea as what @tarekelgindy suggested above:
Would it be possible to implement a "nested dictionary like" structure to store the objects in your opinion? 
By that I mean that when creating a line for instance

```python
from ditto.models.line import Line
api_line = Line(model)
api_line.name = "new_line"
```

this line will be stored implicitly in ```model.models["lines"]``` or ```model.models.lines``` such that we can recover it through:

```python
model.set_names()
new_line_obj = model.models.lines["new_line"]
```

Or something like that...

*Note: This would mean that a lot of the code in post-processing would have to be changed.*

NicolasGensollen, 2018-06-21:
- [ ] **Step 1:** Update ```Store.py``` and ```Base.py``` to enable this functionality. Basically when the object type and the name are both specified we should have unicity. When we query the model with only the name, the output should be all objects with this name. @kdheepak 

- [ ] **Step 2:** Update the parsers and post-processing pieces  @tarekelgindy and @NicolasGensollen

NicolasGensollen, 2018-10-10:
@tarekelgindy, looks like we never really did anything here... 
Could this be a possible task for Bhavya?