NatLabRockies/ditto #68: OpenDSS resulting file is incorrect

jdlara-berkeley, 2018-04-30:
I have been testing the conversion from CYME to OpenDSS and solving the power flow problem, the results from the power flow are only for the source bus. After testing with several CYME files including the 13 bus system data_files in the repository. 

The lines needed to reproduce the problem are as follows: 

```Python 
m = Store()
folder = 'IEEE13'
r = Reader(data_folder_path=folder)     
r.parse(m, verbose = True)
w = Writer(output_path=folder)
w.write(m)
dss.run_command("clear")
dss.run_command('redirect {master}'.format(master=os.path.join(folder,'master.dss')))
dss.run_command('Solve')
dss.run_command('show voltages LN Nodes')
```
I have made the tests with several CYME folders and the problem is the same, the output voltages area as follows and are consistent with the output of the command ```dss.Bus.puVmagAngle()```
 
```
LINE-GROUND and LINE-LINE VOLTAGES BY BUS & NODE

Bus     Node    VLN (kV)   Angle      pu     Base kV    Node-Node   VLL (kV)  Angle      pu

650_SRC 1       2.4018 /_    0.0    0.3336    12.470        1-2       4.16 /_   30.0    0.3336
   -    2       2.4018 /_ -120.0    0.3336    12.470        2-3       4.16 /_  -90.0    0.3336
   -    3       2.4018 /_  120.0    0.3336    12.470        3-1       4.16 /_  150.0    0.3336
633 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
634 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
7 ..... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
RG60 .. 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
611 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
675 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
632 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
652 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
692 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
645 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
646 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
671 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
684 ... 1            0 /_    0.0         0     0.208
650 ... 1            0 /_    0.0         0     0.208        1-2          0 /_    0.0         0
   -    2            0 /_    0.0         0     0.208        2-3          0 /_    0.0         0
   -    3            0 /_    0.0         0     0.208        3-1          0 /_    0.0         0
```

NicolasGensollen, 2018-05-01:
Looks like I declared this solved too fast... 
The fix implemented in #71 improves things but the voltages still drop to 0 after the regulator (instead of the source before #71). 
After a quick look, it seems that there is an issue with the lines declared as ```[OVERHEAD BYPHASE SETTING]``` since none of them make it all the way to ```Lines.dss```.

NicolasGensollen, 2018-05-02:
With fix #73 the voltages shouldn't drop to 0 anymore. 
They are still looking weird so I opened another issue (#74) because this is probably not caused by a bug in the code but more likely by the translation issue for one or more objects.

jdlara-berkeley, 2018-05-04:
The issue is fixed when there is only a source, already validated it with other cyme data files. If I find more issues will re-open the issue.

closed 2018-05-04 by jdlara-berkeley