NatLabRockies/ditto #74: CYME --> OpenDSS validation IEEE 13 bus

NicolasGensollen, 2018-05-02:
When converting the IEEE 13 bus from Cyme to OpenDSS and running power flow, I get weird voltages:

```
LINE-GROUND and LINE-LINE VOLTAGES BY BUS & NODE

Bus  Node    VLN (kV)   Angle      pu     Base kV    Node-Node   VLL (kV)  Angle      pu

650  1        2.515 /_    2.7   0.34933    12.470        1-2     4.1613 /_   30.0    0.3337
   - 2       2.2448 /_ -119.1   0.31179    12.470        2-3     4.1623 /_  -90.0   0.33379
   - 3       2.4568 /_  116.4   0.34124    12.470        3-1     4.1621 /_  150.0   0.33377
633  1       2.9904 /_   -0.2   0.41536    12.470        1-2     4.5338 /_   23.7   0.36357
   - 2       2.1709 /_ -122.3   0.30153    12.470        2-3     4.6587 /_  -89.9   0.37359
   - 3       3.0574 /_  112.5   0.42467    12.470        3-1     5.0358 /_  145.7   0.40383
634  1      0.34504 /_   -0.2     1.245     0.480        1-2    0.52312 /_   23.7    1.0898
   - 2      0.25048 /_ -122.3   0.90384     0.480        2-3    0.53753 /_  -89.9    1.1199
   - 3      0.35277 /_  112.5    1.2729     0.480        3-1    0.58104 /_  145.7    1.2105
7 .. 1        2.515 /_    2.7   0.34933    12.470        1-2     4.1613 /_   30.0    0.3337
   - 2       2.2448 /_ -119.1   0.31179    12.470        2-3     4.1624 /_  -90.0   0.33379
   - 3       2.4569 /_  116.4   0.34125    12.470        3-1     4.1621 /_  150.0   0.33377
RG60 1       2.9863 /_   -0.2   0.41479    12.470        1-2     4.5303 /_   23.8   0.36329
   - 2        2.172 /_ -122.3   0.30169    12.470        2-3     4.6528 /_  -89.8   0.37312
   - 3       3.0517 /_  112.6   0.42388    12.470        3-1     5.0309 /_  145.8   0.40344
611  1            0 /_    0.0         0    12.470        1-2          0 /_    0.0         0
   - 2            0 /_    0.0         0    12.470        2-3     3.0648 /_  -67.6   0.24577
   - 3       3.0648 /_  112.4   0.42569    12.470        3-1     3.0648 /_  112.4   0.24577
675  1       2.9958 /_   -0.3   0.41611    12.470        1-2     4.5393 /_   23.6   0.36402
   - 2       2.1708 /_ -122.4   0.30152    12.470        2-3     4.6672 /_  -90.0   0.37427
   - 3       3.0649 /_  112.3   0.42571    12.470        3-1     5.0433 /_  145.6   0.40443
632  1       2.9904 /_   -0.2   0.41536    12.470        1-2     4.5338 /_   23.7   0.36357
   - 2       2.1709 /_ -122.3   0.30153    12.470        2-3     4.6587 /_  -89.9   0.37359
   - 3       3.0574 /_  112.5   0.42467    12.470        3-1     5.0358 /_  145.7   0.40383
652  1       2.9943 /_   -0.3   0.41591    12.470        1-2     2.9943 /_   -0.3   0.24012
   - 2            0 /_    0.0         0    12.470        2-3          0 /_    0.0         0
   - 3            0 /_    0.0         0    12.470        3-1     2.9943 /_  179.7   0.24012
692  1       2.9945 /_   -0.3   0.41592    12.470        1-2     4.5378 /_   23.6    0.3639
   - 2       2.1704 /_ -122.4   0.30146    12.470        2-3     4.6656 /_  -90.0   0.37415
   - 3       3.0639 /_  112.4   0.42557    12.470        3-1     5.0416 /_  145.6    0.4043
645  1            0 /_    0.0         0     0.208        1-2     2.1708 /_   57.7    10.436
   - 2       2.1708 /_ -122.3    18.076     0.208        2-3     4.6586 /_  -89.9    22.397
   - 3       3.0573 /_  112.5    25.459     0.208        3-1     3.0573 /_  112.5    14.699
646  1            0 /_    0.0         0     0.208        1-2     2.1707 /_   57.7    10.436
   - 2       2.1707 /_ -122.3    18.076     0.208        2-3     4.6585 /_  -89.9    22.397
   - 3       3.0573 /_  112.5    25.458     0.208        3-1     3.0573 /_  112.5    14.698
671  1       2.9944 /_   -0.3   0.41591    12.470        1-2     4.5377 /_   23.6   0.36389
   - 2       2.1703 /_ -122.4   0.30145    12.470        2-3     4.6655 /_  -90.0   0.37414
   - 3       3.0638 /_  112.4   0.42556    12.470        3-1     5.0415 /_  145.6   0.40429
684  1       2.9944 /_   -0.3   0.41591    12.470
   - 3       3.0644 /_  112.4   0.42564    12.470        3-1     5.0417 /_  145.6    0.4043
680  1       2.9944 /_   -0.3   0.41591    12.470        1-2     4.5377 /_   23.6   0.36389
   - 2       2.1703 /_ -122.4   0.30145    12.470        2-3     4.6655 /_  -90.0   0.37414
   - 3       3.0638 /_  112.4   0.42556    12.470        3-1     5.0415 /_  145.6   0.40429
```

To reproduce this, run the following script (make sure you pulled the latest code from Master):

```python
from ditto.readers.cyme.read import Reader
from ditto.writers.opendss.write import Writer
from ditto.store import Store
import opendssdirect as dss

#Create store object
m = Store()

#Read the IEEE 13 node feeder from Cyme
r = Reader(data_folder_path = './tests/data/small_cases/cyme/ieee_13node')
r.parse(m)

#Write to OpenDSS
w = Writer(output_path='./')
[... 40 lines cut ...]
New Line.l684_611 Units=mi Length=0.056818165482742 bus1=684.3 bus2=611.3 switch=n phases=1 Linecode=1P_OH_Code2
New Line.l632_671 Units=mi Length=0.3787878858742 bus1=632.1.2.3 bus2=671.1.2.3 switch=n phases=3 Linecode=3P_OH_Code3
New Line.l671_684 Units=mi Length=0.056818164239999996 bus1=671.1.3 bus2=684.1.3 switch=n phases=2 Linecode=2P_OH_Code4
New Line.l650_632 Units=mi Length=0.3787878858742 bus1=RG60.1.2.3 bus2=632.1.2.3 switch=n phases=3 Linecode=3P_OH_Code3
New Line.l632_633 Units=mi Length=0.0946969404 bus1=632.1.2.3 bus2=633.1.2.3 switch=n phases=3 Linecode=3P_OH_Code5
New Line.l632_645 Units=mi Length=0.0946969404 bus1=632.2.3 bus2=645.2.3 switch=n phases=2 Linecode=2P_OH_Code4
New Line.l645_646 Units=mi Length=0.056818164239999996 bus1=645.2.3 bus2=646.2.3 switch=n phases=2 Linecode=2P_OH_Code4
New Line.l671_680 Units=mi Length=0.1893938808 bus1=671.1.2.3 bus2=680.1.2.3 switch=n phases=3 Linecode=3P_OH_Code3
New Line.l671_692_sw Units=mi Length=0.0 bus1=671.1.2.3 bus2=692.1.2.3 switch=y phases=3
New Line.296 Units=mi Length=0.0 bus1=650.1.2.3 bus2=7.1.2.3 switch=y phases=3

New Linecode.1P_OH_Code0 units=m nphases=1 Rmatrix=(0.0008341910000000001 ) Xmatrix=(3.1839e-07 ) normamps=1000.0
New Linecode.3P_OH_Code1 units=m nphases=3 Rmatrix=(0.000495978 1.0625e-05 1.0625e-05 | 1.0625e-05 0.000490324 1.0625e-05 | 1.0625e-05 1.0625e-05 0.000495978 ) Xmatrix=(0.00027731800000000003 1.0625e-05 1.0625e-05 | 1.0625e-05 0.00025109599999999997 1.0625e-05 | 1.0625e-05 1.0625e-05 0.00027731800000000003 ) normamps=1000.0
New Linecode.1P_OH_Code2 units=m nphases=1 Rmatrix=(0.0008259181987907887 ) Xmatrix=(0.0008372744660767112 ) normamps=230.0
New Linecode.3P_OH_Code3 units=m nphases=3 Rmatrix=(0.0002097450918638558 9.69031722383374e-05 9.537134155492083e-05 | 9.69031722383374e-05 0.0002153849947194483 9.818083599001767e-05 | 9.53713415549208e-05 9.818083599001766e-05 0.0002121816860277862 ) Xmatrix=(0.0006515680602606371 0.00031172618486736406 0.00023918739028285838 | 0.00031172618486736406 0.0006330069364578198 0.00026324338393803325 | 0.00023918739028285838 0.00026324338393803325 0.0006435047518967452 ) normamps=730.0
New Linecode.2P_OH_Code4 units=m nphases=2 Rmatrix=(0.0008225683076791017 0.000128355830585524 | 0.00012835583058552397 0.0008260605613539615 ) Xmatrix=(0.0008431391574044316 0.00028529739480183494 | 0.00028529739480183494 0.0008370259267382621 ) normamps=230.0
New Linecode.3P_OH_Code5 units=m nphases=3 Rmatrix=(0.00046202279694472265 9.69031722383374e-05 9.537134155492083e-05 | 9.69031722383374e-05 0.0004676626998003151 9.818083599001767e-05 | 9.53713415549208e-05 9.818083599001766e-05 0.0004644593911086531 ) Xmatrix=(0.0007526320828721274 0.00031172618486736406 0.00023918739028285838 | 0.00031172618486736406 0.00073407095906931 0.00026324338393803325 | 0.00023918739028285838 0.00026324338393803325 0.0007445687745082355 ) normamps=340.0

Calcvoltagebases

Buscoords buscoords.dss

Solve

```

NicolasGensollen, 2018-05-04:
Looking at the reference OpenDSS model and this output, it looks like we have the following differences (the list is probably non-exhaustive and some differences come from different modeling of the system between OpenDSS and CYME):

### Source

- reference has a 115kV sourcebus connected to a 115/4.16 transformer (sourcebus/650)
- output starts at bus 650 (4.16kV)
- Bus 7 exists in output but not in reference

### Transformers

| object name        | reference           |Output  |
|:------------- |:-------------:|:-----:|
| Reg 1,2,3      | KVA=1666 | KVA=150 |
|  -     | %loadloss=0.01      | %loadloss is absent |
| - | XHL=0.01      |   XHL is absent |

### Regulators

| object name        | reference           |Output  |
|:------------- |:-------------:|:-----:|
| Reg 1,2,3      | vreg=122 | vreg=116.9 |
|  -     | R=3      | R absent |
| - | X=9      |  X absent |

### Loads

- All the output loads use default ```conn``` (Y)
- All the output loads are set as 3 phase
- All the output loads have ```model=1```
- ```load_632``` exists in output but not in reference
- ```load_670``` exists in reference but not in output

### Lines

TODO

### Linecodes

TODO

NicolasGensollen, 2018-05-04:
The ```basekV``` is not set in the output file which explains the crazy per unit values. I'm still not sure why ```12.47``` was used as ```basekV```. When adding ```Set Voltagebases=[4.16, .48]``` to the output file, the results are much less crazy (but still far from acceptable...):

|           |         |         |           |        |         |        | 
|-----------|---------|---------|-----------|--------|---------|--------| 
| Reference |         |         |           | Output |         |        | 
|  pu1_x    |  pu2_x  |  pu3_x  | Bus       |  pu1_y |  pu2_y  |  pu3_y | 
| 0.99997   | 0.99999 | 0.99995 | SOURCEBUS |        |         |        | 
| 0.99991   | 0.99997 | 0.99993 | 650       | 1.0471 | 0.93463 | 1.0229 | 
| 1.056     | 1.0374  | 1.056   | RG60      | 1.2434 | 0.90434 | 1.2706 | 
| 1.0113    | 1.027   | 1.0015  | 633       | 1.2451 | 0.90386 | 1.273  | 
| 0.98716   | 1.0084  | 0.98244 | 634       | 1.245  | 0.90384 | 1.2729 | 
| 0.9828    | 1.0403  | 0.96487 | 671       | 1.2467 | 0.90363 | 1.2756 | 
| 1.0197    | 1.0023  | 0       | 645       | 0      | 0.90381 | 1.2729 | 
| 1.018     | 1.0002  | 0       | 646       | 0      | 0.90379 | 1.2729 | 
| 0.9828    | 1.0403  | 0.96487 | 692       | 1.2468 | 0.90365 | 1.2757 | 
| 0.97627   | 1.0426  | 0.96293 | 675       | 1.2473 | 0.90382 | 1.2761 | 
| 0.96082   | 0       | 0       | 611       | 0      | 0       | 1.276  | 
| 0.97533   | 0       | 0       | 652       | 1.2467 | 0       | 0      | 
| 1.004     | 1.0319  | 0.98974 | 670       |        |         |        | 
| 1.0143    | 1.0289  | 1.0042  | 632       | 1.2451 | 0.90386 | 1.273  | 
| 0.9828    | 1.0403  | 0.96487 | 680       | 1.2467 | 0.90363 | 1.2756 | 
| 0.98087   | 0.96284 | 0       | 684       | 1.2467 | 1.2759  | 0      | 
|           |         |         | 7         | 1.0472 | 0.93463 | 1.0229 |

jdlara-berkeley, 2018-05-31:
The conversion for some single - phase lines is producing 0.0 impedance values. 
From the Cyme file I have these two lines 

```
[OVERHEADLINEUNBALANCED SETTING]
FORMAT_OVERHEADLINEUNBALANCEDSETTING=SectionID,DeviceNumber,LineCableID,Length,ConnectionStatus,CoordX,CoordY,HarmonicModel,FlowConstraintActive,FlowConstraintUnit,MaximumFlow
255,255,4035,11.948835,0,,,2,,,
257,257,4035,362.540187,0,,,2,,,
```
The unbalanced line 257 is correctly created, but not the 255. The resulting Lines.dss file shows the following 

```
New Linecode.1P_OH_Code1 units=m nphases=1 Rmatrix=(0.0 ) Xmatrix=(0.0 )

New Linecode.1P_OH_Code2 units=m nphases=1 Rmatrix=(0.00058 ) Xmatrix=(4.1799999999999996e-07 ) normamps=155.470001

```

NicolasGensollen, 2018-05-31:
@jdlara-berkeley, are you doing CYME --> OpenDSS for the IEEE 13 node? I cannot find these lines in the network file ```tests/data/small_cases/cyme/ieee_13node```. What I have is:

```
[OVERHEADLINEUNBALANCED SETTING]
FORMAT_OVERHEADLINEUNBALANCEDSETTING=SectionID,DeviceNumber,LineCableID,Length,ConnectionStatus,CoordX,CoordY,HarmonicModel,FlowConstraintActive,FlowConstraintUnit,MaximumFlow
L684_652,L684_652,LINE607,243.840000,0,,,2,,,
L692_675,L692_675,LINE606,152.400000,0,,,2,,,
```
And the output is:

*Lines.dss:*
```
New Line.l684_652 Units=mi Length=0.15151510464 bus1=684.1 bus2=652.1 switch=n phases=1 Linecode=1P_OH_Code0

New Line.l692_675 Units=mi Length=0.0946969404 bus1=692.1.2.3 bus2=675.1.2.3 switch=n phases=3  Linecode=3P_OH_Code1
```

*LineCodes.dss:*
```
New Linecode.1P_OH_Code0 units=m nphases=1 Rmatrix=(0.0008341910000000001 ) Xmatrix=(3.1839e-07 ) normamps=1000.0

New Linecode.3P_OH_Code1 units=m nphases=3 Rmatrix=(0.000495978 1.0625e-05 1.0625e-05 | 1.0625e-05 0.000490324 1.0625e-05 | 1.0625e-05 1.0625e-05 0.000495978 ) Xmatrix=(0.00027731800000000003 1.0625e-05 1.0625e-05 | 1.0625e-05 0.00025109599999999997 1.0625e-05 | 1.0625e-05 1.0625e-05 0.00027731800000000003 ) normamps=1000.0
```

Just looking at the two lines you have, they indeed share the same equipment ID but one of them (255) is way shorter than the other. I'm wondering if this could be a rounding error when computing the impedance, but it's just a guess. 
If you tell me what system you are considering I could have a better look.

jdlara-berkeley, 2018-05-31:
@NicolasGensollen It is not the IEEE 13 Node, I have been working with non-public data and Ditto + OpenDSS. 

After checking the OpenDSS data for this issue realized that the problem is on the conversion. I will increase the length and test again.

NicolasGensollen, 2018-06-01:
@jdlara-berkeley, thanks for the files. 
I had a look and I believe the problem comes from the files, not from ditto. 
I know the data is not public and I only display the problematic line in this comment. Let me know if you're not OK with that and I'll delete it.

The overhead lines are defined as:

```
[OVERHEADLINEUNBALANCED SETTING]
FORMAT_OVERHEADLINEUNBALANCEDSETTING=SectionID,DeviceNumber,LineCableID,Length,ConnectionStatus,CoordX,CoordY,HarmonicModel,FlowConstraintActive,FlowConstraintUnit,MaximumFlow
BNA1_255,BNA1_255,4035,11.948835,0,,,2,,,
257,257,4035,362.540187,0,,,2,,,
```

And the equipment ```4035``` is the following:

```
[LINE UNBALANCED]
FORMAT_LINEUNBALANCED=ID,CondID_A,CondID_B,CondID_C,CondID_N,SpacingID,Ra,Rb,Rc,Xa,Xb,Xc,Ba,Bb,Bc,AmpsA,AmpsB,AmpsC,AmpsA_1,AmpsB_1,AmpsC_1,AmpsA_2,AmpsB_2,AmpsC_2,AmpsA_3,AmpsB_3,AmpsC_3,AmpsA_4,AmpsB_4,AmpsC_4,LockImpedance,ShuntConductanceA,ShuntConductanceB,ShuntConductanceC,MutualResistanceAB,MutualResistanceBC,MutualResistanceCA,MutualReactanceAB,MutualReactanceBC,MutualReactanceCA,MutualShuntSusceptanceAB,MutualShuntSusceptanceBC,MutualShuntSusceptanceCA,MutualShuntConductanceAB,MutualShuntConductanceBC,MutualShuntConductanceCA,Favorite,Comments
4035,35AA,NONE,NONE,NONE,MRT,0.580000,0.000000,0.000000,0.418000,0.000000,0.000000,2.092850,0.000000,0.000000,155.470001,0.000000,0.000000,155.470001,0.
```

So the impedance is 0 for phases B and C and different from 0 on phase A. Nonetheless, the section for this line is defined as a phase B section:

```
[SECTION]
FORMAT_SECTION=SectionID,FromNodeID,FromNodeIndex,ToNodeID,ToNodeIndex,Phase,ZoneID,SubNetworkId,EnvironmentID
FORMAT_FEEDER=NetworkID,HeadNodeID,CoordSet,Year,Description,Color,LoadFactor,LossLoadFactorK,Group1,Group2,Group3,TagText,TagProperties,TagDeltaX,TagDeltaY,TagAngle,TagAlignment,TagBorder,TagBackground,TagTextColor,TagBorderColor,TagBackgroundColor,TagLocation,TagFont,TagTextSize,TagOffset,Version,EnvironmentID
FEEDER=BNA,240_HEAD,1,0,,0,1.000000,0.150000,,,,NULL,,,,,,,,,,,,,,,-1,0
BNA1_255,BNA1_253,0,BNA1_255,0,B,,,0
```

So DiTTo is simply assuming that line ```BNA1_255``` is a single phase, phase B line with a 0 impedance as defined in the equipment file. If you change B in A in the section you get a non zero impedance for this line in the OpenDSS output:

```
New Line.bna1_255 Units=mi Length=0.007424659552785 bus1=BNA1_253.1 bus2=BNA1_255.1 switch=n phases=1 Linecode=1P_OH_Code1

New Linecode.1P_OH_Code1 units=m nphases=1 Rmatrix=(0.00058 ) Xmatrix=(4.1799999999999996e-07 ) normamps=155.470001
```

Let me know if this solves the issue or not.