NatLabRockies/ditto #86: ditto command already exists on MacOS

NicolasGensollen, 2018-05-16:
Hey @kdheepak,

I believe other people will probably have this issue on Macs so it's probably better to post it here.
I changed my terminal configuration a couple days ago and I realized that all CLI tests were failing. I tried running a conversion with the CLI and I got the following:

```bash
$ ditto convert --from glm --input ./tests/data/big_cases/gridlabd/ieee_123node/123_node.glm --to dss --output .

ditto: unrecognized option `--from'
Usage: ditto [ <options> ] src [ ... src ] dst
```

```bash
$ ditto --help   
                                                                                                                                                 
Usage: ditto [ <options> ] src [ ... src ] dst

    <options> are any of:
    -h                         print full usage
    -v                         print a line of status for each source copied
    -V                         print a line of status for every file copied
    -X                         do not descend into directories with a different device ID

    -c                         create an archive at dst (by default CPIO format)
    -x                         src(s) are archives
    -z                         gzip compress CPIO archive
    -j                         bzip2 compress CPIO archive
    -k                         archives are PKZip
    --keepParent               parent directory name src is embedded in dst_archive
    --arch archVal             fat files will be thinned to archVal
                               multiple -arch options can be specified
                               archVal should be one of "ppc", "i386", etc
    --bom bomFile              only objects present in bomFile are copied
    --norsrc                   don't preserve resource data
    --noextattr                don't preserve extended attributes
    --noqtn                    don't preserve quarantine information
    --noacl                    don't preserve ACLs
    --sequesterRsrc            copy resources via polite directory (PKZip only)
    --nocache                  don't use filesystem cache for reads/writes
    --hfsCompression           compress files at destination if appropriate
    --nopreserveHFSCompression don't preserve HFS+ compression when copying files
    --zlibCompressionLevel num use compression level 'num' when creating a PKZip archive
    --password                 request password for reading from encrypted PKZip archive
```

It turns out the following ```ditto``` command is used: [ditto](http://osxdaily.com/2014/06/11/use-ditto-copy-files-directories-mac-command-line/) instead of the cli we implemented.

Do you have an idea about this?

kdheepak, 2018-05-16:
That's unfortunate. Should we rename the command line to `ditto-cli`?

NicolasGensollen, 2018-05-16:
Yes, it breaks my heart. ```ditto-cli``` is fine I think.

kdheepak, 2018-05-16:
I did see that `ditto` already existed, but didn't think too many people would miss it. I did not consider PATH issues though. If your Python installation's bin directory e.g. `/Users/$USER/miniconda3/bin` is ahead of `/usr/bin`, then you should never have this issue. For reference, you can type `echo $PATH` to see where your Python bin directory lies relative to `/usr/bin`. If `/usr/bin` is ahead or people are using system Python on their mac, they are likely to run into this issue. If the PR referenced above is merged, then people can always use `ditto-cli` regardless of their PATH settings.

NicolasGensollen, 2018-05-16:
Yes, you're right about the PATH settings. 
Once #87 is merged, we can close this issue I think.

closed 2018-05-16 by kdheepak

bpalmintier, 2018-05-17:
Shoot. I wonder if a different name than ditto-cli would work, especially since cli is rather cryptic. One more drastic option might be to go with the modified spelling of “dito” or “dtto” or similar. Of these dtto keeps the full distribution transformation tool letters. We could even (begrudingly) rebrand things to use that everywhere. Alternatively, perhaps something that is more domain relevant such as ditto-tool or ditto-process or disttran or even ditto-do or ditto-run or dTransform

Closed #86<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNREL%2Fditto%2Fissues%2F86&data=02%7C01%7Cbryan.palmintier%40nrel.gov%7Cfd6d96f706a04d23e66708d5bb782312%7Ca0f29d7e28cd4f5484427885aee7c080%7C0%7C0%7C636621047031430635&sdata=0Yz%2BqiIpTuOzt9zBhP8jvksMS3RdKVfAo6blakDsJkI%3D&reserved=0> via #87<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNREL%2Fditto%2Fpull%2F87&data=02%7C01%7Cbryan.palmintier%40nrel.gov%7Cfd6d96f706a04d23e66708d5bb782312%7Ca0f29d7e28cd4f5484427885aee7c080%7C0%7C0%7C636621047031430635&sdata=JNs4PaHXjHdIjiQCIDng2cpkfSbGEzev45L0I6qhXIA%3D&reserved=0>.

kdheepak, 2018-05-17:
`{name}-cli` is an extremely common pattern for command line interface tools, even in Python. I even found some examples from the first page of search results on Github for me [1], [2], [3]. All of them use `{name}-cli` pattern, and I don't think this will be confusing for users, especially if it is documented the way we have. Additionally, `ditto` will also work in the command line currently and will continue working as long as the users' PATH is configured like most users' PATH would be. We can add a deprecation warning for this if we don't want users to use this going forward. I'm not really a fan of misspelling/shortening the name as it can make it harder to remember and easier to mistype. All that said, I'm fine with adding `ditto-run` if people feel strongly about it (even though I can't find any other Python package that has `{name}-run` in the first few pages of GitHub search results).

[1] - https://github.com/jackzampolin/blockstack-cli
[2] - https://github.com/spaceVStab/bitcoin-cli
[3] - https://github.com/philcrump/habrotate/blob/master/cli/setup.py#L4

PR #87 (2018-05-16): Add ditto cli
Fixes #86. 

Adds `ditto-cli` that does the same thing as `ditto`. We can deprecate `ditto` in a future version if required.