NatLabRockies/ditto #105: Small bug with Black autoformat of JSON files

NicolasGensollen, 2018-05-29:
Example JSON configuration file ```./examples/ieee_13node_opendss_input.json```:

```json
{"master_file":"../tests/data/opendss/ieee_13node/master.dss",
 "buscoordinates_file":"../tests/data/opendss/ieee_13node/buscoord.dss"
}
```

After running Black:

```json
}
    "master_file": "../tests/data/opendss/ieee_13node/master.dss",
    "buscoordinates_file": "../tests/data/opendss/ieee_13node/buscoord.dss",
}
```

When trying to read it from Python:

```python
import json

with open('./examples/ieee_13node_opendss_input.json', 'r') as fp:
    data = json.load(fp)
```

you get the following error caused by the trailing comma:

```python
JSONDecodeError: Expecting property name enclosed in double quotes: line 4 column 1 (char 157)
```

kdheepak, 2018-05-29:
Black thinks the file is a Python dict, and added a comma to the end. That shouldn't have happened. I'll look into it.

NicolasGensollen, 2018-05-29:
@kdheepak, I might have raised this too fast. This is what I get when I open this JSON file and run Black as a vim plugin before closing it.
I don't think the pre-commit hook will do this if you change the json file and try to re-commit it.
So, #103 shouldn't cause this kind of problems.

NicolasGensollen, 2018-05-29:
Yes, the pre-commit ignores the file as expected.
Sorry for the "fake issue" notifications...

closed 2018-05-29 by NicolasGensollen

kdheepak, 2018-05-29:
No problem!