NatLabRockies/ditto #110: Default Values

NicolasGensollen, 2018-06-05:
This is something we definitely have to do at some point. Currently, almost everything defaults to ```None```.

The easiest way would be to set static default values for equipments based on the most common equipments we found in utility data.

A better way would be to update the default values based on what was provided by the user. This probably means having an internal representation of equipments against which we could compare a new object. We could then use the closest known equipment to fill in the missing values.

tarekelgindy, 2018-06-05:
This is a big issue we need to address. I'd like to allow the user to select whether or not they want to use suggested default values from a small library that we include within DiTTo.

NicolasGensollen, 2018-10-10:
The LA100 conversion work will probably need this capability, which means pretty soon... 
Changing the priority and assigning my self.