NatLabRockies/ditto #117: Add unit tests and validation tests

NicolasGensollen, 2018-06-07:
**Priority 1**
Basic unit tests checking that values are parsed correctly (this shouldn't be too hard to implement)

**Priority 2**
Tests performing power flows on the input and output and compare them. Ideally, we shouldn't allow any merge that worsen the quality of a conversion (should be possible for OpenDSS, seems much more complicated for other formats)

NicolasGensollen, 2018-06-18:
Todo list for milestone First Validation:

- [ ] Automatic validation script
- [ ] Need to solve #65 to make sure we can compare power flows
- [ ] Run the power flows and plot histograms

bpalmintier, 2018-10-11:
Can we separate this into test vs. validation a separate issues.

NicolasGensollen, 2018-10-11:
Validation is now in a separate issue (#193).