NatLabRockies/ditto #127: Problem with installation

mahi001, 2018-06-20:
Hi, the following code in cmd `pip install git+https://github.nrel.gov/NREL/ditto.git@master` is returning the attached error. 

 Cloning https://github.nrel.gov/NREL/ditto.git (to master) to c:\users\mrahman\appdata\local\temp\pip-f24kzqz8-build
fatal: unable to access 'https://github.nrel.gov/NREL/ditto.git/': Could not resolve host: github.nrel.gov
  Complete output from command C:\Users\mrahman\AppData\Local\GitHubDesktop\app-1.2.3\resources\app\git\cmd\git.exe clone -q https://github.nrel.gov/NREL/ditto.git C:\Users\mrahman\AppData\Local\Temp\pip-f24kzqz8-build:

----------------------------------------
Cleaning up...
Command C:\Users\mrahman\AppData\Local\GitHubDesktop\app-1.2.3\resources\app\git\cmd\git.exe clone -q https://github.nrel.gov/NREL/ditto.git C:\Users\mrahman\AppData\Local\Temp\pip-f24kzqz8-build failed with error code 128 in None
Storing debug log for failure in C:\Users\mrahman\pip\pip.log

Thanks - 
Mahi, EnerNex

kdheepak, 2018-06-20:
@mahi001 Can you change the command line to the following and try again?

```bash
pip install git+https://github.com/NREL/ditto.git@master
```

NicolasGensollen, 2018-06-20:
Hi,
Could you try using 

```bash
$ pip install git+https://github.com/NREL/ditto.git@master
```
instead of:

```bash
$ pip install git+https://github.nrel.gov/NREL/ditto.git@master
```

which is pointing to the previous internal NREL repo. 

On a side note, it looks like ```installation.md``` needs some updates (at least for a few lines...).

kdheepak, 2018-06-20:
I made the change in 41da698e4973db5329cca930aaa416ee80a04e20 and pushed it directly to master.

mahi001, 2018-06-20:
After updating the installation link I am getting the following error 

`TypeError: Type str doesn't support the buffer API

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in C:\Users\mrahman\AppData\Local\Temp\pip-h7hhwyyb-build
Storing debug log for failure in C:\Users\mrahman\pip\pip.log`

P.S. My python version is 3.4.1, and pip version is 1.5.6 and git version 2.17.1

kdheepak, 2018-06-20:
Hmm. Can you share the full stacktrace? Or share the `pip.log` file. I'm able to install it in my Python 3.4.5 installation.

kdheepak, 2018-06-20:
I'm testing this with Python 3.4.1 and find the same error message. Full traceback is below:

```
$ pip install -e ".[dev,test]"
Obtaining file:///Users/$USER/GitRepos/ditto
    Complete output from command python setup.py egg_info:
    running egg_info
    creating ditto.egg-info
    writing ditto.egg-info/PKG-INFO
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/Users/$USER/GitRepos/ditto/setup.py", line 114, in <module>
        cmdclass={"develop": PostDevelopCommand},
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/site-packages/setuptools-27.2.0-py3.4.egg/setuptools/command/egg_info.py", line 188, in run
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/site-packages/setuptools-27.2.0-py3.4.egg/setuptools/command/egg_info.py", line 403, in write_pkg_info
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/distutils/dist.py", line 1108, in write_pkg_info
        self.write_pkg_file(pkg_info)
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/site-packages/setuptools-27.2.0-py3.4.egg/setuptools/dist.py", line 53, in write_pkg_file
      File "/Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/distutils/util.py", line 470, in rfc822_escape
        lines = header.split('\n')
    TypeError: Type str doesn't support the buffer API

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /Users/$USER/GitRepos/ditto/
You are using pip version 9.0.1, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

```

kdheepak, 2018-06-20:
Once PR #128 and PR #129 are merged, this issue will be resolved.

kdheepak, 2018-06-20:
@mahi001 Can you try again now? It should work.

mahi001, 2018-06-20:
The error shows up a lot later this time and as below @kdheepak  @NicolasGensollen . I can send you the log file as well. 

`C:\Users\mrahman>pip install git+https://github.com/NREL/ditto.git@master
Downloading/unpacking git+https://github.com/NREL/ditto.git@master
  Cloning https://github.com/NREL/ditto.git (to master) to c:\users\mrahman\appdata\local\temp\pip-sh0jdxa8-build
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip-sh0jdxa8-build\setup.py) egg_info for package from git+https://github.com/NREL/ditto.git@master
Downloading/unpacking aenum (from ditto==0.1.0)
  Downloading aenum-2.1.2-py3-none-any.whl
Downloading/unpacking click (from ditto==0.1.0)
Downloading/unpacking croniter (from ditto==0.1.0)
  Downloading croniter-0.3.24.tar.gz
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\croniter\setup.py) egg_info for package croniter
    file src\croniter.py (for module croniter) not found
    warning: no files found matching '*.txt'
    warning: no previously-included files matching '*pyc' found anywhere in distribution
Downloading/unpacking funcsigs (from ditto==0.1.0)
  Downloading funcsigs-1.0.2-py2.py3-none-any.whl
Downloading/unpacking future (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\future\setup.py) egg_info for package future
    warning: no files found matching '*.au' under directory 'tests'
    warning: no files found matching '*.gif' under directory 'tests'
    warning: no files found matching '*.txt' under directory 'tests'
Downloading/unpacking jinja2 (from ditto==0.1.0)
Downloading/unpacking lxml (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\lxml\setup.py) egg_info for package lxml
    Building lxml version 4.2.1.
    Building without Cython.
    ERROR: b"'xslt-config' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n"
    ** make sure the development packages of libxml2 and libxslt are installed **

    Using build configuration of libxslt
    warning: no previously-included files found matching '*.py'
Downloading/unpacking networkx (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\networkx\setup.py) egg_info for package networkx
    warning: no files found matching '*.html' under directory 'doc'
    warning: no files found matching '*.css' under directory 'doc'
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '.svn' found anywhere in distribution
    no previously-included directories found matching 'doc\build'
    no previously-included directories found matching 'doc\auto_examples'
    no previously-included directories found matching 'doc\modules'
    no previously-included directories found matching 'doc\reference\generated'
    no previously-included directories found matching 'doc\reference\algorithms\generated'
    no previously-included directories found matching 'doc\reference\classes\generated'
    no previously-included directories found matching 'doc\reference\readwrite\generated'
Downloading/unpacking pandas (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\pandas\setup.py) egg_info for package pandas
    warning: no files found matching 'pyproject.toml'
    no previously-included directories found matching 'doc\build'
    warning: no previously-included files matching '*.so' found anywhere in distribution
    warning: no previously-included files matching '*.pyd' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '#*' found anywhere in distribution
    warning: no previously-included files matching '.git*' found anywhere in distribution
    warning: no previously-included files matching '.DS_Store' found anywhere in distribution
    warning: no previously-included files matching '*.png' found anywhere in distribution
Downloading/unpacking pytest (from ditto==0.1.0)
Downloading/unpacking six (from ditto==0.1.0)
  Downloading six-1.11.0-py2.py3-none-any.whl
Downloading/unpacking traitlets (from ditto==0.1.0)
Downloading/unpacking xlrd (from ditto==0.1.0)
Downloading/unpacking OpenDSSDirect.py (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\OpenDSSDirect.py\setup.py) egg_info for package OpenDSSDirect.py
    warning: no previously-included files matching '__pycache__' found under directory '*'
    warning: no previously-included files matching '*.py[co]' found under directory '*'
    warning: no files found matching '*.jpg' under directory 'docs'
Downloading/unpacking XlsxWriter (from ditto==0.1.0)
Downloading/unpacking json-tricks (from ditto==0.1.0)
[... 54 lines cut ...]
  File "C:\Python34\lib\site-packages\pip\basecommand.py", line 122, in main
    status = self.run(options, args)
  File "C:\Python34\lib\site-packages\pip\commands\install.py", line 302, in run
    requirement_set.cleanup_files(bundle=self.bundle)
  File "C:\Python34\lib\site-packages\pip\req.py", line 1319, in cleanup_files
    req.remove_temporary_source()
[... cut ...]
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python34\lib\shutil.py", line 367, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python34\lib\shutil.py", line 372, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Python34\lib\site-packages\pip\util.py", line 53, in rmtree_errorhandler
    (exctype is PermissionError and value.args[3] == 5) #python3.3
IndexError: tuple index out of range

  Storing debug log for failure in C:\Users\mrahman\pip\pip.log`

kdheepak, 2018-06-20:
Can you try using Miniconda3 instead?  Or do you have to use the standard Python installation. Installing dependencies in standard Python on Windows can be challenging. If you are restricted to using the Python installation you have on that machine let me know. I'll have to spin up a Windows VM with that specific version of Python to see if I can replicate the same issue.

kdheepak, 2018-06-20:
@mahi001 I edited some of the formatting on your comment so that I could read the error more clearly. It seems like there was a permission issue. You might have to resolve that before the installation can proceed.

Can you also try the following two commands and report the output back separately?

```bash
pip install croniter
```

```bash
pip install lxml
```

You can use three backtick marks to fence the output, so that it is formatted appropriately. e.g. below

````
```
output goes here and is formatted as code
```
````

mahi001, 2018-06-21:
Do we need to have C++ for this? 

`$ pip install croniter`

```
Requirement already satisfied (use --upgrade to upgrade): croniter in c:\python34\lib\site-packages
Downloading/unpacking python-dateutil (from croniter)
Downloading/unpacking six>=1.5 (from python-dateutil->croniter)
  Downloading six-1.11.0-py2.py3-none-any.whl
Installing collected packages: python-dateutil, six
Successfully installed python-dateutil six
Cleaning up...
```
...........................................................
`$ pip install lxml`
```
Downloading/unpacking lxml
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\lxml\setup.py) egg_info for package lxml
    Building lxml version 4.2.1.
    Building without Cython.
    ERROR: b"'xslt-config' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n"
    ** make sure the development packages of libxml2 and libxslt are installed **

    Using build configuration of libxslt
    warning: no previously-included files found matching '*.py'
Installing collected packages: lxml
  Running setup.py install for lxml
    Building lxml version 4.2.1.
    Building without Cython.
    ERROR: b"'xslt-config' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n"
    ** make sure the development packages of libxml2 and libxslt are installed **

    Using build configuration of libxslt
    building 'lxml.etree' extension
    error: Microsoft Visual C++ 10.0 is required. Get it with "Microsoft Windows SDK 7.1": www.microsoft.com/download/details.aspx?id=8279
    Complete output from command C:\Python34\python.exe -c "import setuptools, tokenize;__file__='C:\\Users\\mrahman\\AppData\\Local\\Temp\\pip_build_mrahman\\lxml\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record C:\Users\mrahman\AppData\Local\Temp\pip-nj1k4av5-record\install-record.txt --single-version-externally-managed --compile:
    Building lxml version 4.2.1.

Building without Cython.

ERROR: b"'xslt-config' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n"

** make sure the development packages of libxml2 and libxslt are installed **

Using build configuration of libxslt

running install

running build

running build_py

creating build

creating build\lib.win32-3.4

creating build\lib.win32-3.4\lxml

copying src\lxml\builder.py -> build\lib.win32-3.4\lxml

copying src\lxml\cssselect.py -> build\lib.win32-3.4\lxml

copying src\lxml\doctestcompare.py -> build\lib.win32-3.4\lxml

copying src\lxml\ElementInclude.py -> build\lib.win32-3.4\lxml

copying src\lxml\pyclasslookup.py -> build\lib.win32-3.4\lxml

copying src\lxml\sax.py -> build\lib.win32-3.4\lxml

[... 98 lines cut ...]
creating build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

copying src\lxml\isoschematron\resources\xsl\iso-schematron-xslt1\iso_abstract_expand.xsl -> build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

copying src\lxml\isoschematron\resources\xsl\iso-schematron-xslt1\iso_dsdl_include.xsl -> build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

copying src\lxml\isoschematron\resources\xsl\iso-schematron-xslt1\iso_schematron_message.xsl -> build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

copying src\lxml\isoschematron\resources\xsl\iso-schematron-xslt1\iso_schematron_skeleton_for_xslt1.xsl -> build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

copying src\lxml\isoschematron\resources\xsl\iso-schematron-xslt1\iso_svrl_for_xslt1.xsl -> build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

copying src\lxml\isoschematron\resources\xsl\iso-schematron-xslt1\readme.txt -> build\lib.win32-3.4\lxml\isoschematron\resources\xsl\iso-schematron-xslt1

running build_ext

building 'lxml.etree' extension

error: Microsoft Visual C++ 10.0 is required. Get it with "Microsoft Windows SDK 7.1": www.microsoft.com/download/details.aspx?id=8279

----------------------------------------
Cleaning up...
Command C:\Python34\python.exe -c "import setuptools, tokenize;__file__='C:\\Users\\mrahman\\AppData\\Local\\Temp\\pip_build_mrahman\\lxml\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record C:\Users\mrahman\AppData\Local\Temp\pip-nj1k4av5-record\install-record.txt --single-version-externally-managed --compile failed with error code 1 in C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\lxml
Storing debug log for failure in C:\Users\mrahman\pip\pip.log
```

kdheepak, 2018-06-21:
You wouldn't need a C++ compiler if you used Anaconda/Miniconda, but it appears for standard Python 3.4.1 on Windows you need a C++ compiler to install lxml. I've submitted a PR to make lxml optional. Once that PR is merged, you should be able to install it.

kdheepak, 2018-06-21:
@mahi001 can you try again?

mahi001, 2018-06-21:
Installation update: 
Status: Not Installed

`$ pip install git+https://github.com/NREL/ditto.git@master`

```
Downloading/unpacking git+https://github.com/NREL/ditto.git@master
  Cloning https://github.com/NREL/ditto.git (to master) to c:\users\mrahman\appdata\local\temp\pip-ps1oswst-build
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip-ps1oswst-build\setup.py) egg_info for package from git+https://github.com/NREL/ditto.git@master
Requirement already satisfied (use --upgrade to upgrade): aenum in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): click in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): croniter in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): funcsigs in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): future in c:\python34\lib\site-packages (from ditto==0.1.0)
Downloading/unpacking jinja2 (from ditto==0.1.0)
Downloading/unpacking networkx (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\networkx\setup.py) egg_info for package networkx
    warning: no files found matching '*.html' under directory 'doc'
    warning: no files found matching '*.css' under directory 'doc'
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '.svn' found anywhere in distribution
    no previously-included directories found matching 'doc\build'
    no previously-included directories found matching 'doc\auto_examples'
    no previously-included directories found matching 'doc\modules'
    no previously-included directories found matching 'doc\reference\generated'
    no previously-included directories found matching 'doc\reference\algorithms\generated'
    no previously-included directories found matching 'doc\reference\classes\generated'
    no previously-included directories found matching 'doc\reference\readwrite\generated'
Downloading/unpacking pandas (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\pandas\setup.py) egg_info for package pandas
    warning: no files found matching 'pyproject.toml'
    no previously-included directories found matching 'doc\build'
    warning: no previously-included files matching '*.so' found anywhere in distribution
    warning: no previously-included files matching '*.pyd' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '#*' found anywhere in distribution
    warning: no previously-included files matching '.git*' found anywhere in distribution
    warning: no previously-included files matching '.DS_Store' found anywhere in distribution
    warning: no previously-included files matching '*.png' found anywhere in distribution
Downloading/unpacking pytest (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): six in c:\python34\lib\site-packages (from ditto==0.1.0)
Downloading/unpacking traitlets (from ditto==0.1.0)
Downloading/unpacking xlrd (from ditto==0.1.0)
Downloading/unpacking OpenDSSDirect.py (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\OpenDSSDirect.py\setup.py) egg_info for package OpenDSSDirect.py
    warning: no previously-included files matching '__pycache__' found under directory '*'
    warning: no previously-included files matching '*.py[co]' found under directory '*'
    warning: no files found matching '*.jpg' under directory 'docs'
Downloading/unpacking XlsxWriter (from ditto==0.1.0)
Downloading/unpacking json-tricks (from ditto==0.1.0)
  Downloading json_tricks-3.12.1.tar.gz
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\json-tricks\setup.py) egg_info for package json-tricks
Downloading/unpacking scipy (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): numpy>=1.13.0 in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in c:\python34\lib\site-packages (from croniter->ditto==0.1.0)
Downloading/unpacking MarkupSafe>=0.23 (from jinja2->ditto==0.1.0)
  Downloading MarkupSafe-1.0.tar.gz
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\MarkupSafe\setup.py) egg_info for package MarkupSafe
Downloading/unpacking decorator>=4.1.0 (from networkx->ditto==0.1.0)
  Downloading decorator-4.3.0-py2.py3-none-any.whl
Downloading/unpacking pytz>=2011k (from pandas->ditto==0.1.0)
Downloading/unpacking py>=1.5.0 (from pytest->ditto==0.1.0)
Downloading/unpacking attrs>=17.4.0 (from pytest->ditto==0.1.0)
  Downloading attrs-18.1.0-py2.py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): setuptools in c:\python34\lib\site-packages (from pytest->ditto==0.1.0)
Downloading/unpacking more-itertools>=4.0.0 (from pytest->ditto==0.1.0)
Downloading/unpacking atomicwrites>=1.0 (from pytest->ditto==0.1.0)
  Downloading atomicwrites-1.1.5-py2.py3-none-any.whl
[... 55 lines cut ...]
[... cut ...]
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python34\lib\shutil.py", line 367, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python34\lib\shutil.py", line 372, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Python34\lib\site-packages\pip\util.py", line 53, in rmtree_errorhandler
    (exctype is PermissionError and value.args[3] == 5) #python3.3
IndexError: tuple index out of range

  Storing debug log for failure in C:\Users\mrahman\pip\pip.log
```

kdheepak, 2018-06-21:
Can you type `where pip` and tell us the results. I'm also on Gitter, if you want to chat there.

mahi001, 2018-06-21:
`$ where pip`
```
C:\Python34\Scripts\pip.exe
```

kdheepak, 2018-06-21:
This is probably not going to solve the issue, but I'm curious what will happen if you try the following. From your previous log, it seemed like the pip was using an old installation (web cache?). The following line will point it to the latest commit.

```
pip install git+https://github.com/NREL/ditto.git@c4daec5
```

mahi001, 2018-06-21:
Update: 
Status: Not installed
Number of attempts: 2

First Attempt: 
`pip install git+https://github.com/NREL/ditto.git@c4daec5`
```
Downloading/unpacking git+https://github.com/NREL/ditto.git@c4daec5
  Cloning https://github.com/NREL/ditto.git (to c4daec5) to c:\users\mrahman\appdata\local\temp\pip-yd7ezbcn-build
  Could not find a tag or branch 'c4daec5', assuming commit.
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip-yd7ezbcn-build\setup.py) egg_info for package from git+https://github.com/NREL/ditto.git@c4daec5
Requirement already satisfied (use --upgrade to upgrade): aenum in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): click in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): croniter in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): funcsigs in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): future in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): jinja2 in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): networkx in c:\python34\lib\site-packages (from ditto==0.1.0)
Downloading/unpacking pandas (from ditto==0.1.0)

pip can't proceed with requirement 'pandas (from ditto==0.1.0)' due to a pre-existing build directory.
 location: C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\pandas
This is likely due to a previous installation that failed.
pip is being responsible and not assuming it can delete this.
Please delete it and try again.

Storing debug log for failure in C:\Users\mrahman\pip\pip.log
```
Manually deleted the ...\pip_build_mrahman folder and attempted 2nd time 

Second Attempt: 

`$ pip install git+https://github.com/NREL/ditto.git@c4daec5`
```
Downloading/unpacking git+https://github.com/NREL/ditto.git@c4daec5
  Cloning https://github.com/NREL/ditto.git (to c4daec5) to c:\users\mrahman\appdata\local\temp\pip-s5o2543i-build
  Could not find a tag or branch 'c4daec5', assuming commit.
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip-s5o2543i-build\setup.py) egg_info for package from git+https://github.com/NREL/ditto.git@c4daec5
Requirement already satisfied (use --upgrade to upgrade): aenum in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): click in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): croniter in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): funcsigs in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): future in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): jinja2 in c:\python34\lib\site-packages (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): networkx in c:\python34\lib\site-packages (from ditto==0.1.0)
Downloading/unpacking pandas (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\pandas\setup.py) egg_info for package pandas
    warning: no files found matching 'pyproject.toml'
    no previously-included directories found matching 'doc\build'
    warning: no previously-included files matching '*.so' found anywhere in distribution
    warning: no previously-included files matching '*.pyd' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '#*' found anywhere in distribution
    warning: no previously-included files matching '.git*' found anywhere in distribution
    warning: no previously-included files matching '.DS_Store' found anywhere in distribution
    warning: no previously-included files matching '*.png' found anywhere in distribution
Downloading/unpacking pytest (from ditto==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): six in c:\python34\lib\site-packages (from ditto==0.1.0)
Downloading/unpacking traitlets (from ditto==0.1.0)
Downloading/unpacking xlrd (from ditto==0.1.0)
Downloading/unpacking OpenDSSDirect.py (from ditto==0.1.0)
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\OpenDSSDirect.py\setup.py) egg_info for package OpenDSSDirect.py
    warning: no previously-included files matching '__pycache__' found under directory '*'
    warning: no previously-included files matching '*.py[co]' found under directory '*'
    warning: no files found matching '*.jpg' under directory 'docs'
Downloading/unpacking XlsxWriter (from ditto==0.1.0)
Downloading/unpacking json-tricks (from ditto==0.1.0)
  Downloading json_tricks-3.12.1.tar.gz
  Running setup.py (path:C:\Users\mrahman\AppData\Local\Temp\pip_build_mrahman\json-tricks\setup.py) egg_info for package json-tricks
[... 2091 lines cut ...]
    status = self.run(options, args)
  File "C:\Python34\lib\site-packages\pip\commands\install.py", line 302, in run
    requirement_set.cleanup_files(bundle=self.bundle)
  File "C:\Python34\lib\site-packages\pip\req.py", line 1319, in cleanup_files
[... cut ...]
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python34\lib\shutil.py", line 367, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python34\lib\shutil.py", line 372, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Python34\lib\site-packages\pip\util.py", line 53, in rmtree_errorhandler
    (exctype is PermissionError and value.args[3] == 5) #python3.3
IndexError: tuple index out of range

  Storing debug log for failure in C:\Users\mrahman\pip\pip.log
```

kdheepak, 2018-06-21:
DiTTo currently uses Pandas, NumPy and SciPy, which are all Python dependencies that depend on C/C++/Fortran code. We can make a version of DiTTo available that does not use these dependencies, but then there will be some features you will not have access to. I highly recommend installing and using `conda` (See [Anaconda](https://www.anaconda.com/download/)/[Miniconda](https://conda.io/miniconda.html)) since that will resolve all installation problems for you. If you don't want to or cannot use `conda`, I'd recommend upgrading your Python installation to Python3.6, since it is likely that you will have pre-built binaries for Pandas, NumPy and SciPy available for you to install through pip. If you absolutely have to use Python3.4, let us know and we'll figure out a way to make DiTTo available for you to install without access to all the features.

kdheepak, 2018-06-22:
Once #140 is merged, core DiTTo will be close to a pure Python package you will be able to install the core of DiTTo easily. However, to actually use it you'll need some of the dependencies that the plugins are using. This means you'll still need to install Pandas/NumPy/SciPy depending on what readers, writers or metrics you require.

kdheepak, 2018-06-22:
@mahi001 You can try again now, the default installation should give you the core pure Python DiTTo. Additional installation requirements can be added using the extras.

e.g. default pure Python core DiTTo

```bash
pip install git+https://github.com/NREL/ditto.git@master#egg=ditto
```

Additional dependencies

```bash
pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[all]
# OR
pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[opendss,gridlabd]
```

Full list of options for extras can be seen [here](https://github.com/NREL/ditto/blob/54296f4/setup.py#L111-L124).

kdheepak, 2018-06-22:
In my shell, I need to make sure that the last argument is in double quotes (`"`):

e.g.

```bash
pip install "git+https://github.com/NREL/ditto.git@master#egg=ditto[all]"
```

PR #129 (2018-06-20): Use encoding in open for README.md
Fixes #127
PR #134 (2018-06-21): Make lxml dependency optional
Fixes #127