NatLabRockies/ditto #130: Tests failing in Python 3.4.x

kdheepak, 2018-06-20:
Test failure log below:

```
============================================================================================================== FAILURES ===============================================================================================================
________________________________________________________________________________________________________ test_cyme_to_ephasor _________________________________________________________________________________________________________

    def test_cyme_to_ephasor():
        '''
            Test the Cyme to Ephasor conversion.
        '''
        from ditto.store import Store
        from ditto.readers.cyme.read import Reader
        from ditto.writers.ephasor.write import Writer

        cyme_models=[f for f in os.listdir(os.path.join(current_directory, 'data/small_cases/cyme/')) if not f.startswith('.')]
        for model in cyme_models:
            m = Store()
            r = Reader(data_folder_path=os.path.join(current_directory, 'data/small_cases/cyme',model))
>           r.parse(m)

tests/test_cyme_to_ephasor.py:33:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
ditto/readers/cyme/read.py:810: in parse
    super(Reader, self).parse(model, **kwargs)
ditto/readers/abstract_reader.py:480: in parse
    s = self.parse_lines(model)
ditto/readers/cyme/read.py:3433: in parse_lines
    distance_matrix, gmr_list, resistance_list
ditto/readers/abstract_reader.py:431: in get_primitive_impedance_matrix
    primitive_impedance_matrix[-1].append(self.carson_equation_self(r_list[i], GMR_list[i]))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <ditto.readers.cyme.read.Reader object at 0x10f536f98>, ri = None, GMRi = None

    def carson_equation_self(self, ri, GMRi):
        '''Carson's equation for self impedance.'''

>       return complex(ri + .0953, .12134 * (np.log(1.0 / GMRi) + 7.93402))
E       TypeError: unsupported operand type(s) for +: 'NoneType' and 'float'

ditto/readers/abstract_reader.py:415: TypeError
---------------------------------------------------------------------------------------------------------- Captured log call ----------------------------------------------------------------------------------------------------------
read.py                    850 WARNING  Warning. The current CYME--->DiTTo reader was developed with documentation of CYME 8.0. Your version is 5.04. You might want to check the differences between the two.
read.py                    850 WARNING  Warning. The current CYME--->DiTTo reader was developed with documentation of CYME 8.0. Your version is 8.00. You might want to check the differences between the two.
________________________________________________________________________________________________________ test_cyme_to_gridlabd ________________________________________________________________________________________________________

    def test_cyme_to_gridlabd():
        '''
            Test the Cyme to GridlabD conversion.
        '''
        from ditto.store import Store
        from ditto.readers.cyme.read import Reader
        from ditto.writers.gridlabd.write import Writer

        cyme_models=[f for f in os.listdir(os.path.join(current_directory, './data/small_cases/cyme/')) if not f.startswith('.')]
        for model in cyme_models:
            m = Store()
            r = Reader(data_folder_path=os.path.join(current_directory, './data/small_cases/cyme',model))
>           r.parse(m)

tests/test_cyme_to_gridlabd.py:33:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
ditto/readers/cyme/read.py:810: in parse
    super(Reader, self).parse(model, **kwargs)
ditto/readers/abstract_reader.py:480: in parse
    s = self.parse_lines(model)
ditto/readers/cyme/read.py:3433: in parse_lines
    distance_matrix, gmr_list, resistance_list
ditto/readers/abstract_reader.py:431: in get_primitive_impedance_matrix
    primitive_impedance_matrix[-1].append(self.carson_equation_self(r_list[i], GMR_list[i]))
[... 136 lines cut ...]

>       return complex(ri + .0953, .12134 * (np.log(1.0 / GMRi) + 7.93402))
E       TypeError: unsupported operand type(s) for +: 'NoneType' and 'float'

ditto/readers/abstract_reader.py:415: TypeError
---------------------------------------------------------------------------------------------------------- Captured log call ----------------------------------------------------------------------------------------------------------
read.py                    850 WARNING  Warning. The current CYME--->DiTTo reader was developed with documentation of CYME 8.0. Your version is 5.04. You might want to check the differences between the two.
read.py                    850 WARNING  Warning. The current CYME--->DiTTo reader was developed with documentation of CYME 8.0. Your version is 8.00. You might want to check the differences between the two.
========================================================================================================== warnings summary ===========================================================================================================
tests/test_cyme_to_ephasor.py::test_cyme_to_ephasor
  /Users/$USER/GitRepos/ditto/ditto/readers/cyme/read.py:3415: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
    idx_to_remove = np.argwhere(gmr_list == None).flatten()

tests/test_json_parsers.py::test_opendss_to_json
  /Users/$USER/miniconda3/envs/ditto-env-py3.4.1/lib/python3.4/site-packages/opendssdirect/utils.py:98: UserWarning: Pandas cannot be installed. Please see documentation for how to install extra dependencies.
    "Pandas cannot be installed. Please see documentation for how to install extra dependencies."

tests/test_json_parsers.py::test_cyme_to_json
  /Users/$USER/GitRepos/ditto/ditto/models/base.py:28: UserWarning: Duplicate name 650 being set. Object overwritten.
    warnings.warn("Duplicate name %s being set. Object overwritten." % name)

-- Docs: http://doc.pytest.org/en/latest/warnings.html
===================================================================================== 5 failed, 25 passed, 2 skipped, 3 warnings in 20.18 seconds =====================================================================================

```

NicolasGensollen, 2018-06-20:
Hmmm, this looks like issue #22 that we solved a long time ago.
I don't understand why you get this error.

Could you tell me what example system is causing this? (I guess CYME IEEE 4 node).

When I add the following to the abstract reader:

```python
def carson_equation_self(self, ri, GMRi):
        '''Carson's equation for self impedance.'''
        if ri is None:
            raise ValueError("Resistance is None. Cannot compute Carson's equation")
        if GMRi is None:
            raise ValueError("GMR is None. Cannot compute Carson's equation")
        if GMRi == 0:
            raise ValueError("GMR is zero. Cannot compute Carson's equation")
        return complex(ri + .0953, .12134 * (np.log(1.0 / GMRi) + 7.93402))
```

I don't have any error when parsing the IEEE 4 node and 13 node. Could you try on your side?

NicolasGensollen, 2018-06-20:
@kdheepak, I'm wondering if this could be linked to the version of Numpy you are using when running these tests. The line 

```python
idx_to_remove = np.argwhere(gmr_list == None).flatten()
```

might be the source of the problem here.

kdheepak, 2018-06-20:
Interesting. I upgraded to `1.14.5` and now the tests all pass. Great! Thanks for identifying the issue so quickly. It might be worth adding to the documentation. Also, if we are using numpy we should add it to the requirements install_requires, even if we know it will be installed as part of pandas or other dependencies. We can then ensure we get a specific version of higher of numpy.

```
install_requires = [
*install_requires,
"numpy>=1.13.0",
]
```

NicolasGensollen, 2018-06-20:
I agree for explicitly adding Numpy to the requirements. 
I approved #132 which should solve this issue.

closed 2018-06-20 by kdheepak