NatLabRockies/ditto #138: Make DiTTo Conda installable

NicolasGensollen, 2018-06-21:


kdheepak, 2018-08-03:
We need releases on GitHub for this to work. Can you make a tagged release?

kdheepak, 2018-08-03:
I made it such that you can `conda install -c kdheepak ditto`. See https://anaconda.org/kdheepak/ditto. 

Right now the only test I'm doing is checking is the python package imports, i.e. `import ditto`. I'm not checking if all functionality works correctly. This also assumes that the current latest master is a release, and that is what is being packaged; i.e. this commit hash: a6d5cacb30429cfb03d44c455d44d09d314dee4b

Once we make a tagged release and run a full test suite on Windows, Mac and Linux I'll update the package to point to that instead. I'll also make it available on conda-forge instead of my channel.

NicolasGensollen, 2018-08-04:
Thanks @kdheepak for looking into this! I'll see early next week with @tarekelgindy which PR we want to merge and which bugs we want to solve before making a release.

kdheepak, 2018-08-04:
Great! There's one branch that I pushed that I'd like merged before tagging.

NicolasGensollen, 2018-08-06:
I'd really like to merge #152, #154, and #162 before tagging. 
Also, fixing issues #157 and #160 would be nice.
@tarekelgindy, let me know what your thoughts are on this.