NatLabRockies/ditto #147: Pre-commit hooks not working with python environment

tarekelgindy, 2018-07-20:
@kdheepak 
I seem to have problems getting the black pre-commit hooks to work properly with my virtual environment on Peregrine. Any idea what the issue here might be?

> An unexpected error has occurred: CalledProcessError: Command: 
>('/home/telgindy/anaconda2/envs/py36/bin/python', '-mvirtualenv', '/home/telgindy/.cache/pre-commit/repojjkyn0u8/py_env-python3.6', '-p', 'python3.6')
Return code: 100
Expected return code: 0
Output:
    Using base prefix '/home/telgindy/anaconda2/envs/py36'
    New python executable in /home/telgindy/.cache/pre-commit/repojjkyn0u8/py_env-python3.6/bin/python3.6
    Also creating executable in /home/telgindy/.cache/pre-commit/repojjkyn0u8/py_env-python3.6/bin/python
    ERROR: The executable /home/telgindy/.cache/pre-commit/repojjkyn0u8/py_env-python3.6/bin/python3.6 is not functioning
    ERROR: It thinks sys.prefix is '/' (should be '/home/telgindy/.cache/pre-commit/repojjkyn0u8/py_env-python3.6')
    ERROR: virtualenv is not compatible with this system or executable
    Running virtualenv with interpreter /home/telgindy/anaconda2/envs/py36/bin/python3.6
>Errors:
    /home/telgindy/.cache/pre-commit/repojjkyn0u8/py_env-python3.6/bin/python3.6: error while loading shared libraries: libpython3.6m.so.1.0: cannot open shared object file: No such file or directory

kdheepak, 2018-07-20:
Can you use conda environments instead of `virtualenv`?

tarekelgindy, 2018-07-20:
I believe that I currently am. The environment I'm using is located here:

> /home/telgindy/anaconda2/envs/py36

and was created using

> conda create -n py36 python=3.6 anaconda

it's activated using 

> source activate py36

kdheepak, 2018-07-20:
Can you try the following?

```
conda create --prefix=/home/telgindy/local/conda-envs/python36 python=3.6
source activate /home/telgindy/local/conda-envs/python36
```

tarekelgindy, 2018-07-20:
We solved this in a pretty ugly way.
Firstly 
`conda install virtualenv`
`conda install pre_commit -c conda-forge`

As per this issue here: https://github.com/pre-commit/pre-commit/issues/701

This uses a version of virtualenv that's compatible with conda. 

We then had an issue with pip in virtualenv. It was configured to use pip 10 but we had to downgrade it to pip 9 as per this issue: https://github.com/pypa/pip/issues/5267

We ended up putting the pip 9 wheel from here:

https://github.com/GeoNode/geonode-win-installer/blob/master/python_deps/pip-9.0.1-py2.py3-none-any.whl

into here:
` /home/telgindy/anaconda2/envs/py36/lib/python3.6/site-packages/virtualenv_support/` on my machine.

Once we did that we had pip 9 being used by virtualenv to install packages that it needed and were able to run it properly within pre-commit

closed 2018-07-20 by tarekelgindy

zhihaoy, 2020-01-30:
It does not work with `python_venv` language.  This is problematic for team working on different platforms because `python_venv` has better compatibility.

kdheepak, 2020-02-06:
Can you expand on why it doesn't work with `venv`?