NatLabRockies/ditto #158: Synergi --> OpenDSS impedance matrices are off

NicolasGensollen, 2018-08-01:
Hi @xzhu1991,

Pete and I are trying to convert some feeders from Synergi to OpenDSS and it seems like the impedance matrices in the output are off. 

Looking at the reader code, it looks like this is where the conversion between sequence impedances and phase matrices takes place:

https://github.com/NREL/ditto/blob/5a7c5029320fa2d97a16618436a6f8e6ed91b959/ditto/readers/synergi/read.py#L702-L757

I think there are a few things that we should change (but I might be wrong since I don't know Synergi):

- The ```1/3``` coefficient seems to be missing
- The ```10^-3``` coefficient is probably not right. We need to convert the sequence values to ```Ohms per meters``` since DiTTo is in meters, so we assume here that the values are given in kilometers. 

I looked a little bit in the tables and most attributes end with ```per_LUL```, ```per_MUL```, or ```per_SUL```. I assume these mean ```large unit length```, ```medium unit length```, and ```small unit length``` but this is just a guess. Do you know more about this?

Also, the table ```SAI_Equ_Control``` seems to have a field called ```LengthUnits``` (set to ```English2``` in my case). I looked at the Synergi software with Pete and I believe that you can have three possibilities:

- *Metric*: ```LUL=km```, ```MUL=m```, ```SUL=mm```
- *English1*: ```LUL=kft```, ```MUL=ft```, ```SUL=mft```
- *English2*: ```LUL=miles```, ```MUL=??```, ```SUL=??```

Again, this is just a guess... Do you know more on this?

xzhu1991, 2018-08-01:
> e6ed91b959/ditto/readers/synergi/read.py#L702-L757

I could not open Synergi on my laptop. If you open Synergi, you should be
able to see the length unit of the lines.

I asked Aadil before about the length unit. It should be mile.

I updated the coefficient on my new code. But I haven't push them to github
yet.

The correct one should be: coeff = 1/4827 , basically from mile to meter (
1/1609), and multiply 1/3.

NicolasGensollen, 2018-08-01:
Thanks for your answer @xzhu1991 

This is the solution I came up with for now (see ff6d8a90adbb78ff647b1650b9b8798edb392767):

```python
# If LengthUnits is set to English2 or not defined , then assume miles
if LengthUnits == "English2" or LengthUnits is None:
    coeff = 0.000621371
# Else, if LengthUnits is set to English1, assume kft
elif LengthUnits == "English1":
    coeff = 3.28084 * 10 ** -3
# Else, if LengthUnits is set to Metric, assume km
elif LengthUnits == "Metric":
    coeff = 10 ** -3
else:
    raise ValueError("LengthUnits <{}> is not valid.".format(LengthUnits))

coeff *= 1.0 / 3.0
```

I still need to make sure the conversion is not introducing other errors in the impedance matrices.

NicolasGensollen, 2018-08-06:
Hi @xzhu1991, I'm not sure I understand this code in the Synergi reader when you compute the impedance matrix for a two phase line:

```python
if NPhase == 2:
    impedance_matrix = [[coeff * complex(float(r0), float(x0))]]
```

It seems like ```r1``` and ```x1``` are not used.