NatLabRockies/ditto #177: Underground lines not parser

jd-lara, 2018-08-17:
While trying to parse CYME files with an underground line the OpenDSS output doesn't include the line. 
The file are located [here](https://github.com/jd-lara/TTA_BahiaNegra/tree/master/BN), and in the folder DSS_files you can find the resulting OpenDSS output where in Lines.dss line `BNA1_260` doesn't show up. 

The code for the conversion is [here](https://github.com/jd-lara/TTA_BahiaNegra/blob/master/Ditto_CYMEtoDSS.py)

NicolasGensollen, 2018-08-17:
Hey @jd-lara,

Looking at the equipement file, the concentric neutral cables are defined under the header ```[CONCENTRIC NEUTRAL CABLE]``` while the cyme reader looks for them under the header ```[CABLE CONCENTRIC NEUTRAL]``` which is the header specified in the the CYME documentation v.8.0 rev.1-0.

So, if you replace ```[CONCENTRIC NEUTRAL CABLE]``` by ```[CABLE CONCENTRIC NEUTRAL]``` in your equipment file, the conversion should work.

Let me know if this solves the issue. If not, I'll have a closer look.

NicolasGensollen, 2018-08-17:
@jd-lara, I submitted a PR (#178) which should solve this in a better way than forcing the user to edit the ASCII files. 
Once merged, you'll be able to use ```[CONCENTRIC NEUTRAL CABLE]``` or ```[CABLE CONCENTRIC NEUTRAL]```.

jd-lara, 2018-08-17:
Thanks, I will update and test.

NicolasGensollen, 2018-08-21:
@jd-lara, #178 has been merged. 
This should work as is without any modification on the ASCII files.

jd-lara, 2018-09-17:
The fix works, thank you.

closed 2018-09-17 by jd-lara