NatLabRockies/ditto #183: Errors for PNNL Taxonomy Feeders Conversion from glm to DSS

yvelaga, 2018-09-11:
Hi,

I am using the DiTTo tool for the first time. I have installed using pip in Anaconda. There are couple of errors for two feeders I tried to convert using simple python script (attached). Here are those:

1. Heron area is negative for R1-12.47-1.glm
2. Transformer BasekV list has "None" 
3. Heron area is infinity for R1-12.47-2.glm

I tried to solve the first two errors. I have no idea why the max distance is zero for the third error. I am not sure if this is an issue with input files or installation. Please let me know if you need any more information.
[image]

NicolasGensollen, 2018-09-11:
Hi @yvelaga, I'm not sure I understand, did the code crash because of the infinite area or do you see weird things in the DSS output files?
The Python conversion script looks perfectly fine.
I'm tagging @tarekelgindy who knows more than me about the Gridlab-D reader.

tarekelgindy, 2018-09-14:
Hi @yvelaga 

Thanks a lot for raising this issue. This also relates to issue #28. Two of the points that you raised in here were bugs in DiTTo - 2 phase underground lines were not being handled correctly and the OpenDSS writer was not checking for Nonetypes when iterating over transformer windings. I've addressed these in the most recent pull request. 

The second issue that you raised is a bit more delicate. In R1-12.47-2.glm, a spacing that was causing problems was line_spacing:10 which has 4 wires with distance of 0 from each other. To me, this looks like an error in the spacing. However, line_spacing:13 had two wires (Phase A and neutral) with 0 distance. For an underground wire this may mean concentric neutral wires.

There are other taxonomy feeders which have incorrect spacings as well. For example in feeder R5-12.47-1.glm, line_spacing:17 has distances of (20.8, 20.8 and 48.0) between the three wires with phase B, C and N. This configuration isn't physically possible.

My solution for all of these is to create a default line configuration. If the code for computing the line spacing fails for any reason, a warning is printed and a default configuration is used. For overhead wires this is a spacing with the neutral wire at 32 feet high and phase wires spaced 2 feet apart at 30 feet high.  For underground wires this default is all wires at depth 6 feet placed half a foot away from each other.

This doesn't really address the issue of concentric neutral vs. non concentric underground wires, but it should provide a good temporary solution while we address this more fundamental problem. 

I've tested the conversion for all the taxonomy feeders with these fixes and they all seem to convert fine. I haven't tested the powerflow on them though. These changes have been pulled into the master branch. If you update with 
`git pull origin master` 
on the master branch you should have these changes applied to your code.

Let me know if this works and we can close the issue