NatLabRockies/ditto #185: Schema.Json is Missing with Conda Install

yvelaga, 2018-09-20:
Hi @kdheepak, @tarekelgindy 

When I installed ditto using conda, schema.Json is missing from installation files. I have to put it there manually. Recently, when I updated the ditto using pip, file did not get updated.

Please let me know if you have any questions.

kdheepak, 2018-09-20:
The `conda` recipe is waiting on DiTTo to have a tagged release (See reference #138). @NicolasGensollen @tarekelgindy where are we with a tagged release of DiTTo?

NicolasGensollen, 2018-09-20:
@kdheepak, checking with @tarekelgindy right now. We can probably do a `1.0.0-alpha.1` release right now with what we have in master to solve #138 and this issue, even if we still have some bugs we'd like to solve.

tarekelgindy, 2018-09-20:
I agree with that. There's a final pull request I've just added now but honestly we're overdue for a tagged release.

version `1.0.0-alpha.1` sounds good to me.

NicolasGensollen, 2018-09-20:
@kdheepak, just did a release after merging #186.