NatLabRockies/ditto #188: Pandas-access not working on Windows

NicolasGensollen, 2018-09-21:
Looks like reading MDB databases on a Windows machine through pandas-access is not working. So using the Synergi reader on Windows is raising a `FileNotFoundError`...

In other words, doing the following on Windows seems to crash:
```
$ pip install pandas_access
```
```python
>>> import pandas_access as mdb
>>> mdb.list_tables(path_to_mdb_database)
FileNotFoundError: [Errno 2] No such file or directory: 'mdb-tables': 'mdb-tables'
```

A quick and dirty fix could be to have to `db_parser.py`, one for windows using previous code from @AadilLatif and one for Mac OS and redirect the right one in the reader.
@kdheepak did you see this issue before?

kdheepak, 2018-10-02:
`mdb-tables` is a command line tool that needs to exist in order to perform this function. See [here](https://github.com/jbn/pandas_access/blob/3c08bef8e0fc180774e3a6b58b06450c8b6aa15e/pandas_access/__init__.py#L25).

NicolasGensollen, 2018-10-02:
I believe @tarekelgindy was able to get this working on his Windows. I believe he had to install `mdb-tools` manually. Could you confirm this @tarekelgindy? If so, we should probably add that to the dependencies in some way (or at least write some documentation on what to do).

tarekelgindy, 2018-10-02:
The way that i got this working on windows was to use some pre-compiled files located on this repository:
https://github.com/lsgunth/mdbtools-win

I cloned this repo, added the location to the install path and then mbd-tables was able to be run.

The mdbtools repository is located here:
https://github.com/brianb/mdbtools/

and @NicolasGensollen used homebrew to install it on mac. I think you could use linuxbrew or an direct install to run it on linux.

This isn't an ideal situation, but seems to be an ok workaround for now.