NatLabRockies/ditto #202: Unable to use "git commit"

kavuribhavya, 2018-10-19:
When I try to use the command "git commit -m "Message"", I see the below error:

```
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ git commit -m "OpenDSS Line Length : First Draft"
[INFO] Installing environment for https://github.com/ambv/black.
[INFO] Once installed this environment will be reused.
[INFO] This may take a few minutes...
An unexpected error has occurred: CalledProcessError: Command: ('C:\\Users\\bkavuri\\AppData\\Local\\Continuum\\anaconda3\\envs\\py3.6\\python.exe', '-mvirtualenv', 'U:\\.cache\\pre-commit\\repo1da_ak1a\\py_env-python3', '-p', 'C:\\python3\\python.exe')
Return code: 3
Expected return code: 0
Output:
    The path C:\python3\python.exe (from --python=C:\python3\python.exe) does not exist

Errors: (none)

Check the log at U:\/.cache\pre-commit\pre-commit.log
(py3.6)

```

I am using this command for the first time after the ditto installation.

NicolasGensollen, 2018-10-19:
@kdheepak do you have any idea about that? I had a look with @tarekelgindy but we weren't able to solve it...

kdheepak, 2018-10-19:
I recommend installing Miniconda3. And then make sure you have precommit
and black installed, which should happen if you use the correct pip install
invocation.

kdheepak, 2018-10-19:
Oops, I didn't see that you already are using Anaconda. Uninstall your
default Python3 install.

kavuribhavya, 2018-10-19:
I don't have Python3 installed other than the Anaconda's python.

kavuribhavya, 2018-10-19:
1. Installed Anaconda and added ```/c/Users/bkavuri/AppData/Local/Continuum/anaconda3/Scripts``` to PATH
2. Created a python3.6 environment and activated it
3. Installed ditto with ```pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[all]```

The path ```C:\python3\python.exe``` doesn't exist.

kdheepak, 2018-10-19:
Clone this git repository and use the `dev` target instead of `all`. See [here](https://nrel.github.io/ditto/developer-guide/install.html).

kavuribhavya, 2018-10-19:
It gives me the same error,

```
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/new_ditto/ditto (new)
$ git commit -m "Test"
[INFO] Installing environment for https://github.com/ambv/black.
[INFO] Once installed this environment will be reused.
[INFO] This may take a few minutes...
An unexpected error has occurred: CalledProcessError: Command: ('c:\\users\\bkavuri\\appdata\\local\\continuum\\anaconda3\\envs\\py3.6\\python.exe', '-mvirtualenv', 'U:\\.cache\\pre-commit\\repo1da_ak1a\\py_env-python3', '-p', 'C:\\python3\\python.exe')
Return code: 3
Expected return code: 0
Output:
    The path C:\python3\python.exe (from --python=C:\python3\python.exe) does not exist

Errors: (none)

Check the log at U:\/.cache\pre-commit\pre-commit.log
(py3.6)

```

kdheepak, 2018-10-19:
Well that's annoying. Are you using a Python conda environment?

```
$ pip uninstall pre-commit ditto black
$ conda create -n ditto-env python=3 -y
$ source activate ditto-env
$ pip install -e ".[all,dev]"
$ pytest -sv
$ git status
```

Can you try the above and paste the logs here?

kdheepak, 2018-10-19:
Hmm. Did you try committing again? Maybe you can't use 3.7? You may have to use Python3.6. Can you try running the same with explicitly specifying the Python environment?

```bash
$ conda env remove --name ditto-env 
$ conda create -n ditto-env python=3.6 -y
$ source activate ditto-env
$ pip install -e ".[all,dev]"
$ git commit
```

kavuribhavya, 2018-10-19:
Yes, I did try to commit again and got the same error. 

I run everything with specifying the Python version and I still get the same error. Am I missing something ?

```
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ source deactivate ditto-env
(py3.6)
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ conda env remove --name ditto-env

Remove all packages in environment C:\Users\bkavuri\AppData\Local\Continuum\anaconda3\envs\ditto-env:

## Package Plan ##

  environment location: C:\Users\bkavuri\AppData\Local\Continuum\anaconda3\envs\ditto-env

The following packages will be REMOVED:

    certifi:        2018.10.15-py37_0
    pip:            10.0.1-py37_0
    python:         3.7.0-hea74fb7_0
    setuptools:     40.4.3-py37_0
    vc:             14.1-h0510ff6_4
    vs2015_runtime: 14.15.26706-h3a45250_0
    wheel:          0.32.1-py37_0
    wincertstore:   0.2-py37_0

Proceed ([y]/n)? y

(py3.6)
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ conda create -n ditto-env python=3.6 -y
Solving environment: ...working... done

## Package Plan ##

  environment location: C:\Users\bkavuri\AppData\Local\Continuum\anaconda3\envs\ditto-env

  added / updated specs:
    - python=3.6

The following NEW packages will be INSTALLED:

    certifi:        2018.10.15-py36_0
    pip:            10.0.1-py36_0
    python:         3.6.6-hea74fb7_0
    setuptools:     40.4.3-py36_0
    vc:             14.1-h0510ff6_4
    vs2015_runtime: 14.15.26706-h3a45250_0
    wheel:          0.32.1-py36_0
    wincertstore:   0.2-py36h7fe50ca_0

Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
#
# To activate this environment, use
#
#     $ conda activate ditto-env
#
# To deactivate an active environment, use
#
#     $ conda deactivate

(py3.6)
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$  source activate ditto-env
(ditto-env)
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ pip install -e ".[all,dev]"
Obtaining file:///C:/Users/bkavuri/Documents/ditto
Collecting click (from ditto==0.1.0)
  Using cached https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl
Collecting future (from ditto==0.1.0)
Collecting networkx (from ditto==0.1.0)
Collecting six (from ditto==0.1.0)
```

```
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ python --version
Python 3.6.6 :: Anaconda, Inc.
(ditto-env)
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ which python
/c/Users/bkavuri/AppData/Local/Continuum/anaconda3/envs/ditto-env/python
(ditto-env)
```

NicolasGensollen, 2018-10-19:
@kdheepak, probably a stupid idea but what if you install Python3 at the expected location (`C:\python3\python.exe`)? Would it work?

kdheepak, 2018-10-19:
I have the same issue on my Windows machine. I'll explore further. 

The problem is that `black` is trying to use `virtualenv` to create a Python3 environment. It appears that the command that is being generated by black's pre-commit hook does not work. To get around this, you can disable the `black` pre-commit hook. Change the `.pre-commit-config.yaml` file to remove the black section or run the following:

```
pre-commit uninstall
```

Don't commit the changes to .pre-commit-config.yaml if there are any. If you make changes to .pre-commit-config.yaml, you will need to reinstall by doing the folllowing

```
pre-commit install
```

Manually run `black` on a Python file after you have committed it, and commit the changes to the repository under a separate commit.

```
black module.py
git add module.py
git commit -m "Fix formatting in module.py"
```

I'll report back if I find a way to fix this.

kavuribhavya, 2018-10-19:
We tried installing Python3.6 at /C/python3 but we get the following error

```
bkavuri@MININT-LJCCB0O MINGW64 /C/Users/bkavuri/Documents/ditto (OpenDSS_unittest)
$ git commit -m "Test"
[INFO] Initializing environment for https://github.com/pre-commit/pre-commit-hooks.
[INFO] Initializing environment for https://github.com/ambv/black.
[INFO] Installing environment for https://github.com/pre-commit/pre-commit-hooks.
[INFO] Once installed this environment will be reused.
[INFO] This may take a few minutes...
[INFO] Installing environment for https://github.com/ambv/black.
[INFO] Once installed this environment will be reused.
[INFO] This may take a few minutes...
An unexpected error has occurred: CalledProcessError: Command: ('U:\\.cache\\pre-commit\\repo882qp8qt\\py_env-python3\\Scripts\\pip.exe', 'install', '.')
Return code: 2
Expected return code: 0
Output:
    Processing u:\.cache\pre-commit\repo882qp8qt

Errors:
    Exception:
    Traceback (most recent call last):
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\basecommand.py", line 228, in main
        status = self.run(options, args)
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\commands\install.py", line 291, in run
        resolver.resolve(requirement_set)
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\resolve.py", line 103, in resolve
        self._resolve_one(requirement_set, req)
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\resolve.py", line 257, in _resolve_one
        abstract_dist = self._get_abstract_dist_for(req_to_install)
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\resolve.py", line 210, in _get_abstract_dist_for
        self.require_hashes
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\operations\prepare.py", line 324, in prepare_linked_requirement
        abstract_dist.prep_for_dist(finder, self.build_isolation)
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\operations\prepare.py", line 126, in prep_for_dist
        build_requirements, isolate = self.req.get_pep_518_info()
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_internal\req\req_install.py", line 448, in get_pep_518_info
        pp_toml = pytoml.load(f)
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\site-packages\pip\_vendor\pytoml\parser.py", line 10, in load
        return loads(fin.read(), translate=translate, filename=getattr(fin, 'name', repr(fin)))
      File "u:\.cache\pre-commit\repo882qp8qt\py_env-python3\lib\encodings\cp1252.py", line 23, in decode
        return codecs.charmap_decode(input,self.errors,decoding_table)[0]
    UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 867: character maps to <undefined>
    You are using pip version 10.0.1, however version 18.1 is available.
    You should consider upgrading via the 'python -m pip install --upgrade pip' command.

Check the log at U:\/.cache\pre-commit\pre-commit.log
(ditto-env)
```

kdheepak, 2018-10-19:
I don't think that's the way to go for this. We need to submit an issue to black.

NicolasGensollen, 2018-10-19:
@kdheepak, I agree that this is not the path to a real fix, but we were just curious to see what would happen with this little trick.

tarekelgindy, 2018-10-20:
@kdheepak @kavuribhavya I get the same issue as well with a fresh clone. Referring back to #147, it might be possible to fix this by using an older pip version to install that virtualenv uses

tarekelgindy, 2018-10-20:
I think this might work. In your python installation (mine is `/c/Users/telgindy/AppData/Local/Programs/Python/Python37-32/Lib/site-packages/virtualenv_support`
modify the pip wheel file you're using. I'm using the pip wheel version from here: https://github.com/GeoNode/geonode-win-installer/blob/master/python_deps/pip-9.0.1-py2.py3-none-any.whl
but you might want to update this with the pip 18 wheel instead. Even if you update pip, virtualenv will be using the wheel installed in the site-packages folder.

tarekelgindy, 2018-10-20:
I wanted to clarify the steps to fix this issue below

Firstly, when installing ditto, on windows you'll need to download pandoc from here:
https://github.com/jgm/pandoc/releases/tag/2.3.1
as described here:
https://pandoc.org/installing.html

You should install this before cloning the git repo and installing it with
`pip install -e .[dev]` 
as a developer or
`pip install -e .`
as a user
However the problem @kavuribhavya  is describing arises when trying to commit changes with git. Pre-commit tries to get setup but fails due to a bad virtual environment being created.

The problem you've been having is because virtualenv in python comes with pip 10.01. This may download packages into it's virtual environment which are not up to date and that's what's causing problems here.
When I look in my virtualenv python installation which is located here for me on windows:
`C:\Users\telgindy\AppData\Local\Programs\Python\Python37-32\Lib\site-packages\virtualenv_support`
there is a file titled: 
`pip-10.0.1-py2.py3-none-any.whl`
This is the pip wheel file that virtualenv is using to download packages for it's virtual environment, which is required to run black and other programs run by pre-commit. We want to replace this wheel file with a more up-to-date file. Currently the most recent pip wheel is 18.1. 

To get the latest pip wheel run 
`python -m pip wheel --wheel-dir=. pip`
This gets the most recent wheel file for pip. Currently, this should download a file titled
`pip-18.1-py2.py3-none-any.whl`
Delete or backup the file 
`pip-10.0.1-py2.py3-none-any.whl`
from your virtualenv folder (mine is here):
`C:\Users\telgindy\AppData\Local\Programs\Python\Python37-32\Lib\site-packages\virtualenv_support`
Then move the recently downloaded wheel file 
`pip-18.1-py2.py3-none-any.whl`
to this folder.

Now when you do 
`git commit -m "message"`
the correct pip is being used to download packages in the virtual environment. I was able to use this to get pre-commit working correctly.

We might want to find a way of integrating these changes into the install so people don't have these problems in the future. @kdheepak how hard do you think that would be?

PR #208 (2018-10-23): Fix pre-commit windows installation issues with Black
Fixes #202
closed 2018-10-23 by NicolasGensollen