NatLabRockies/ditto #265: single phase load voltage needs to be added

jd-lara, 2018-12-04:
When parsing single phase loads from Cyme to OpenDSS, the tool doesn't add a `kv=` field with the LG voltage. OpenDSS interprets this as a L-L voltage by default leading incorrect results in the PowerFlow calculation in OpenDSS. This might be a mismatch between the default in Cyme and the default in OpenDSS for single phase loads.

tarekelgindy, 2018-12-13:
Thanks for raising this and sorry for the late response on this. We've identified this as an issue previously and I'm adding a fix for this now. This should be as simple as writing the kV value for OpenDSS. I expect we'll have a fix for this included tomorrow.

tarekelgindy, 2018-12-22:
I believe this issue has been addressed in the latest pull request #268.

The problem was that nominal voltages weren't being set by the CYME reader, and hence were being ignored by the opendss writer.

Cyme doesn't include nominal voltages explicitly for each node so we had to calculate them through post-processing. Originally we didn't want to include this in the reader, and allow the user to make these changes in post-processing but I think it makes sense to automatically calculate the nominal voltages as part of the reader.

I've put a pull request for the branch to join it into master but until it's merged you can use the branch "cyme_bugfixes" located here: https://github.com/NREL/ditto/tree/cyme_bugfixes/ (will be deleted once the branch is merged to master)