NatLabRockies/ditto #287: Support for MacOS?

dpinney, 2019-05-01:
Hi, I'm running macOS 10.14 with Python 2.7.x from homebrew. Is this a supported platform? I wasn't able get it working.

When I do the recommended `pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[all]` from the readme.md it errors out with:

```
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/local/README.md'
Consider using the `--user` option or check the permissions.
```

So I added the --user flag and it appeared to install correctly at the user level.

The ditto-cli tool, however, wasn't on the path. And I was not able to locate it among the scripts in the project.

I switched to the python interface and tried:

```python
from ditto.readers.gridlabd.read import Reader
from ditto.writers.opendss.write import Writer
from ditto.store import Store
model = Store()
glm_reader = Reader(master_file = 'IEEE-13.glm')
glm_reader.parse(model)
dss_writer = Writer(output_path=".")
```

Which errors out with:
```
Traceback (most recent call last):
  File "/workdir/1.convert.py", line 11, in <module>
    from ditto.readers.gridlabd.read import Reader
  File "/Users/dpinney/Library/Python/2.7/lib/python/site-packages/ditto/readers/gridlabd/__init__.py", line 5, in <module>
    from .read import Reader as GridLABDReader
  File "/Users/dpinney/Library/Python/2.7/lib/python/site-packages/ditto/readers/gridlabd/read.py", line 37, in <module>
    from ditto.formats.gridlabd import gridlabd
  File "/Users/dpinney/Library/Python/2.7/lib/python/site-packages/ditto/formats/gridlabd/gridlabd.py", line 52, in <module>
    __create()
  File "/Users/dpinney/Library/Python/2.7/lib/python/site-packages/ditto/formats/gridlabd/gridlabd.py", line 15, in __create
    with open(os.path.join(dir_path, "schema.json")) as f:
IOError: [Errno 2] No such file or directory: '/Users/dpinney/Library/Python/2.7/lib/python/site-packages/ditto/formats/gridlabd/schema.json'
```

I'm still keen on using ditto. What platform is supported?

kdheepak, 2019-05-01:
MacOS is supported, that's what I'm using DiTTo on. This seems like the installation didn't occur as expected. 

Can you check if that file exists on your machine?: `/Users/dpinney/Library/Python/2.7/lib/python/site-packages/ditto/formats/gridlabd/schema.json`

I'm guessing it does not. Does that folder exist?

Also, I'd recommend not using Python2, and moving to Python3. Python2's end of life is coming up shortly https://pythonclock.org/, and we'll likely drop support for Python2 in DiTTo as well. If you have to use it now though, it should still all work fine.

kdheepak, 2019-05-01:
I tested this locally and this is a bug. I fixed it in a separate branch. If you want to use it at the moment you can do the following:

```
pip install "git+https://github.com/NREL/ditto.git@kd/fix-missing-json-files#egg=ditto[all]"
```

Alternatively, you can wait till the fix is merged to `master` and use the instructions in the README.

tarekelgindy, 2019-05-02:
This change has now been moved into master. If you re-install it the required schema should now be included.

To address the `permission denied` issue, this has to do with where python is trying to install packages in system folders that don't have base permissions. The `--user` option will install it in a local folder. 
A good description of this can be found here: 
https://github.com/googlesamples/assistant-sdk-python/issues/236#issuecomment-383039470

If you're using anaconda, I would recommend creating an conda environment, which will contain all the packages required for DiTTo and then pip installing once you've activated your custom environment.

dpinney, 2019-05-02:
Fix partially confirmed.

`pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[all]` without `--user` works now, and ditto-cli is on the path.

However, when I run `ditto-cli convert --input "./IEEE-13.glm" --from gridlabd --to opendss --output ./`, I get a similar error about schema.json:

```
Traceback (most recent call last):
  File "/usr/local/bin/ditto-cli", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/ditto/cli.py", line 172, in convert
    registered_reader_class=_load(registered_readers, from_reader_name),
  File "/usr/local/lib/python2.7/site-packages/ditto/cli.py", line 36, in _load
    cls = plugins[name].load()
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2346, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2352, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/site-packages/ditto/readers/gridlabd/__init__.py", line 5, in <module>
    from .read import Reader as GridLABDReader
  File "/usr/local/lib/python2.7/site-packages/ditto/readers/gridlabd/read.py", line 37, in <module>
    from ditto.formats.gridlabd import gridlabd
  File "/usr/local/lib/python2.7/site-packages/ditto/formats/gridlabd/gridlabd.py", line 52, in <module>
    __create()
  File "/usr/local/lib/python2.7/site-packages/ditto/formats/gridlabd/gridlabd.py", line 15, in __create
    with open(os.path.join(dir_path, "schema.json")) as f:
IOError: [Errno 2] No such file or directory: '/usr/local/lib/python2.7/site-packages/ditto/formats/gridlabd/schema.json'
```

Only contents of /usr/local/lib/python2.7/site-packages/ditto/formats/gridlabd/ are:

```
__init__.py
base.py
gridlabd.py
```

dpinney, 2019-05-02:
I added the schema.json to the .../formats/gridlabd/ folder manually and now I get this error:

```
No handlers could be found for logger "ditto.writers.abstract_writer"
Traceback (most recent call last):
  File "/usr/local/bin/ditto-cli", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/ditto/cli.py", line 179, in convert
    remove_opendss_default_values_flag=remove_opendss_default_values_flag,
  File "/usr/local/lib/python2.7/site-packages/ditto/converter.py", line 244, in convert
    self.writer.write(self.m)
  File "/usr/local/lib/python2.7/site-packages/ditto/writers/opendss/write.py", line 191, in write
    s = self.write_regulators(model)
  File "/usr/local/lib/python2.7/site-packages/ditto/writers/opendss/write.py", line 2170, in write_regulators
    str(i.windings[w].nominal_voltage * 10 ** -3)
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```

dpinney, 2019-05-03:
I get the same error when I test against the version of IEEE13 that's [in this repo](https://raw.githubusercontent.com/NREL/ditto/6543462ec6cf441092abd063b1009ff5d4e2eeb6/ditto/readers/gridlabd/13node_simplified.glm), so it looks like this is an install problem.

kdheepak, 2019-05-03:
I'm getting the same error on my machine

```
$ ditto-cli convert --input "./IEEE-13.glm" --from gridlabd --to opendss --output ./

Log file currently not supported, please contact the developers for information on how to generate log files
Failed to read spacing - using default positions
Traceback (most recent call last):
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/bin/ditto-cli", line 11, in <module>
    load_entry_point('ditto', 'console_scripts', 'ditto-cli')()
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/Users/$USER/miniconda3/envs/ditto-env-py3.6/lib/python3.6/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/$USER/GitRepos/ditto/ditto/cli.py", line 179, in convert
    remove_opendss_default_values_flag=remove_opendss_default_values_flag,
  File "/Users/$USER/GitRepos/ditto/ditto/converter.py", line 244, in convert
    self.writer.write(self.m)
  File "/Users/$USER/GitRepos/ditto/ditto/writers/opendss/write.py", line 191, in write
    s = self.write_regulators(model)
  File "/Users/$USER/GitRepos/ditto/ditto/writers/opendss/write.py", line 2170, in write_regulators
    str(i.windings[w].nominal_voltage * 10 ** -3)
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```

@tarekelgindy any idea what might be going on here?

tarekelgindy, 2019-05-03:
I've found out what's causing this - it's a bug with the voltages not being read with the regulators. On a side note, we should probably include testing of the gridlab-d reader in the testing folder to avoid discoveries like this. I'll make these changes and add them to master asap.

tarekelgindy, 2019-05-03:
I think I've fixed this bug. It's on the branch located [here](https://github.com/NREL/ditto/tree/gld_reg_bugfix) which will be merged into master soon. This is quite specific to regulators which is why we might have missed this issue previously. However, this highlights the need for additional testing of the gridlab-d reader..

tarekelgindy, 2019-05-06:
@dpinney  This has now been merged into master. Does this address your issue? 
We'll add a few gridlab-d reader tests to DiTTo to make sure that rest of the test systems are being converted correctly as well.

dpinney, 2019-05-06:
Re-installed just now. Fix confirmed on the 3 models I was trying to convert.

If you're looking for good reader tests, I recommend PNNL's [taxonomic feeders](https://github.com/gridlab-d/Taxonomy_Feeders). Real circuits at full scale, well-validated. I had to remove the complex_assert objects, but after that ditto would convert them.

sssaha, 2019-05-07:
I am facing the issue of missing schema.json file. 
I am using Anaconda in a Linux machine. 
This is what I did : 
`conda create -n ditto python=3.6
conda activate ditto
pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[all]
git clone https://github.com/NREL/ditto.git
cd ditto/examples
python ieee_123node_gridlabd_to_opendss.py
`

The error I am getting is 
`FileNotFoundError: [Errno 2] No such file or directory: '/home/sssaha/anaconda3/envs/ditto/lib/python3.6/site-packages/ditto/formats/gridlabd/schema.json'
`

What did I do wrong?

PR #288 (2019-05-02): Add recursive-include for schema.json
Fixes #287