NatLabRockies/ditto #292: Linux Installtion Issue

sssaha, 2019-05-09:
I am facing the issue of missing schema.json file. 
I am using Anaconda in a Linux machine. 
This is what I did : 
`conda create -n ditto python=3.6`
`conda activate ditto`
`pip install git+https://github.com/NREL/ditto.git@master#egg=ditto[all]`
`git clone https://github.com/NREL/ditto.git`
`cd ditto/examples`
`python ieee_123node_gridlabd_to_opendss.py`

The error I am getting is 
`FileNotFoundError: [Errno 2] No such file or directory: '/home/sssaha/anaconda3/envs/ditto/lib/python3.6/site-packages/ditto/formats/gridlabd/schema.json'
`

What did I do wrong?

tarekelgindy, 2019-05-09:
It looks like the schema isn't getting included in the pip install. @kdheepak do you know why this might be the case?

A short term solution to this is to install ditto from the clone: `pip install -e .` in the base folder of the cloned repository, which should then include the location of the schema file.

sssaha, 2019-05-09:
So, I deleted my old environment, and created a fresh environment, and did ` pip install -e .[all]`, which installed the required packages. 
Then I tried running the `python ieee_123node_gridlabd_to_opendss.py` from the examples folder, which threw the following error: 

`>>> Creating empty model...
>>> Reading from Gridlab-D...
Traceback (most recent call last):
  File "ieee_123node_gridlabd_to_opendss.py", line 59, in <module>
    main()
  File "ieee_123node_gridlabd_to_opendss.py", line 40, in main
    r.parse(model)
  File "/home/sssaha/ditto/ditto/readers/gridlabd/read.py", line 2209, in parse
    from_nominal_voltage = self.all_gld_objects[from_element]["nominal_voltage"]
KeyError: 'node:150'
`