NatLabRockies/ditto #294: Problem in Convering GridlabD model to OpenDSSModel

sssaha, 2019-05-24:
Hello 
I just tried the ditto-cli tool on the IEEE 13 node GridlabD model (I downloaded the model from GridlabD GitHub repository and then removed the recorder). 
I noticed a couple of problems while inspecting the OpenDSS file..
1. The regulators are missing KVA ratings.
2. OpenDSS requires an explicit definition of Slack Bus. If not mentioned, then one bus has to have the name source-bus in the model. This was not present in the OpenDSS model. 
3. The R matrix and X matrix of the lines should have the size that matches with their phase information. For example, for a single phase line, the Rmatrix and Xmatrix should have only one values, not a value padded with zeros. OpenDSS throws that as an error. 
The original file and the converted files can be found here: 
https://www.dropbox.com/sh/izmk96njw17iedk/AAAnaucMjJB0zsdCghqRvbNea?dl=0

kdheepak, 2019-05-29:
Thanks for making the issue and posting the files.

jenna-lynne, 2024-07-29:
Bumping this issue as it has not been resolved. Can you provide the files again? The link says the files have been deleted.