NatLabRockies/ditto #295: Issue Converting Cyme to OpenDSS - Unable to Parse Error

keithdsz, 2019-05-29:
Hi All,

I have an issue converting cyme(sxst) to opendss

I get the following error:
Could not find [SI] or [IMPERIAL] unit system information. Unable to parse.

Any help addressing this error would be greatly appreciated.

Thanks,

Keith

kdheepak, 2019-05-29:
Can you post a screenshot or a full stacktrace?

tarekelgindy, 2019-05-29:
This error will happen when ditto is unable to parse a cyme file (we need to improve the error messaging on this)

DITTo currently converts the ascii representation of a cyme model, and it sounds like you're trying to convert the .sxst format. If you have access to the original cyme model, export the network, load and equipment ascii files and then follow the conversion instructions provided here: https://nrel.github.io/ditto/cli-examples.

keithdsz, 2019-05-30:
@kdheepak 
[image]
Here is the image. 

@tarekelgindy : Thank you. I'll try that.

keithdsz, 2019-05-30:
@tarekelgindy : I tried what you suggested. Now I get the following error. (I converted my sxst file to the three files required)

[image]

ashraf-bem, 2020-05-31:
I am facing the same issue. KeyError: 'positivesequenceresistance'.  I have exported the network, load and equipment ascii files and ran simulations. Was there any solution provided?

linquanbai, 2020-07-22:
Same issue here any update on this?

thomas-baskin, 2021-06-10:
> I am facing the same issue. KeyError: 'positivesequenceresistance'. I have exported the network, load and equipment ascii files and ran simulations. Was there any solution provided?

I currently have this issue. Did you end up figuring it out? @ashraf-bem

keithdsz, 2021-06-10:
I looked at the python code and found a solution, you need to redefine how the source model is written. Let me know if you need help with this.

thomas-baskin, 2021-06-11:
Did you just wrap the positivesequenceresistance code in a try and except clause? There was a commented try and except clause around the positivesequnceresistance line in the code. DiTTo didn't crash and my output looks fairly okay but I'm not quite sure what changed. I basically just ignored the positivesquenceresistance with the try and except. Any thoughts? @keithdsz

keithdsz, 2021-06-12:
@thomas-baskin : I modified the Network.txt file. 

Look for the keyword source equivalent in it, you should see something like this:

FORMAT_SOURCEEQUIVALENT=NodeID,LoadModelName,Voltage,OperatingAngle1,OperatingAngle2,OperatingAngle3,UseSecondLevelImpedance,FirstLevelR1,FirstLevelX1,FirstLevelR0,FirstLevelX0,FirstLevelR2,FirstLevelX2,SecondLevelR1,SecondLevelX1,SecondLevelR0,SecondLevelX0,SecondLevelR2,SecondLevelX2,OperatingVoltage1,OperatingVoltage2,OperatingVoltage3,BaseMVA,ImpedanceUnit
Testfd,,13.090000,0.000000,-120.000000,120.000000,0,0.130567,0.987034,0.121143,2.810794,0.130567,0.987034,0.130567,0.987034,0.121143,2.810794,0.130567,0.987034,7.200000,7.200000,7.200000,0.000000,0

replace the format with something like this:

FORMAT_SOURCEEQUIVALENT=NodeID,Voltage,OperatingAngle1,OperatingAngle2,OperatingAngle3,PositiveSequenceResistance,PositiveSequenceReactance,ZeroSequenceResistance,ZeroSequenceReactance,Configuration,OperatingVoltage1,OperatingVoltage2,OperatingVoltage3,ImpedanceUnit
testfd,13,0,-120,120,0.130567,0.987034,0.121143,2.810794,0,7.2,7.2,7.2,0

CYME has multiple ways of defining it's source, the way DiTTo is coded, it cannot "understand" the former format, so you need to rewrite the definition in the latter format. 

The main items you need to take care of are sequence impedances, voltages and angles. 

The error we're seeing is triggered because DiTTo cannot find the keywords in the Network.txt file to reconstruct the source. As you've pointed out the try/except method will also work as it bypasses this section. 

Hope this helps.