NatLabRockies/ditto #311: Constantly getting "Pandas is not installed. Please see documentation for how to install extra dependencies."

spark706, 2019-11-09:
I tried Anaconda Python 2.7 which comes with Pandas and get the above error.  I tried running Ditto-CLI in a Python 2 environment, using Spyder, Jupyter Notebooks, and VS Code all attempting to simply run the example scripts provided in the Github download.  I am working on the basis that if the example scripts convert then I am using the tool correctly and can then attempt to convert my Cyme models.

What am I missing?

kdheepak, 2019-11-09:
Can you share more information / screenshots / full stacktrace?

spark706, 2019-11-09:
Here are screen shots from my CLI including Python version and installed
packages. These are from running the script directly within the folder
containing the example script and I tried 2 different ones.

spark706, 2019-11-09:
Attached are screen shots from my CLI showing Python version 2.7, installed packages, and two different scripts I attempted directly from within the examples folder.

[image]
[image]
[image]

spark706, 2019-11-10:
I was looking at the code for the opendssdirect/utils.py and I wanted to ask about something.

`from __future__ import absolute_import
import inspect
import warnings

from ._utils import get_string

is_pandas_installed = True

try:
    import pandas as pd
except ImportError:
    is_pandas_installed = False

class Iterator(object):
    def __init__(self, module, function):
        assert inspect.ismodule(module), "{module} must be of type module".format(
            module=module
        )
        self.module = module
        self.function = function

    def __iter__(self):
        self.module.First()

        while True:

            yield getattr(self.module, self.function)

            if not self.module.Next() > 0:
                break`

Why is the program setting `is_pandas_installed`  to `True`, then importing `pandas` and then evaluating twice for `pandas` in the code?

I quickly commented out these, of course breaking other items, but I didn't get an error about `pandas` not being installed.

Just curious.

kdheepak, 2019-11-10:
In ditto, we have [this line](https://github.com/NREL/ditto/blob/b3cb27fac90dd057f96c8862a86442d8bda39a07/ditto/readers/opendss/read.py#L214-L215). That's probably why that warning is generated. 

I'll update OpenDSSDirect.py such that that warning isn't so annoying anymore. For your purposes you can ignore it.

Your other error messages that you shared in the screenshot are not related to Pandas not being installed, it is because of some incorrect data somewhere. I'm pinging @tarekelgindy to see if he has a better idea about what is going on here. It might be related to https://github.com/NREL/ditto/issues/13.

daniel-thom, 2020-08-12:
@kdheepak @tarekelgindy I think we should fix this.  It's a confusing message that happens every time I run a command with an OpenDSS reader.  The warning message is still present in opendssdirect.

It appears that we are setting this flag in order to avoid having a dependency on pandas.  That may have been true at one point in time, but now it is used in many places in DiTTo.  To fix it we just need to make a wrapper function around `dss.utils.class_to_dataframe` that calls `df.to_dict()`.  Alternatively, someone could make a `class_to_dict` method in opendssdirect, which would be very straightforward.

What do you think?  I prefer option 1 above and can take care of it.

kdheepak, 2020-08-13:
I think the most straightforward fix is to add `df.to_dict()`.

PR #343 (2020-08-14): Remove opendssdirect warning about pandas
Implement a dataframe-to-dict converter so that DiTTo can call the
opendssdirect function class_to_dataframe without modifying an internal
variable.

Fixes #311
closed 2020-08-14 by daniel-thom