NatLabRockies/ditto #312: Receiving "TypeError: __init__() takes at least 3 arguments (2 given)"

spark706, 2019-11-09:
Using the following in the ditto command line interface

ditto-cli convert --from cyme --input ieee_123node_cyme_to_opendss.json --to dss --output ./

return the TypeError in the title.  What arg am I missing?

kdheepak, 2019-11-09:
Can you post the full stacktrace?

spark706, 2019-11-09:
Attached is a screen shot from my CLI running the script from within the
examples folder.

spark706, 2019-11-09:
Here is a screen shot of the CLI running the script from within the examples folder.  I also included the Python version used.
[image]

kdheepak, 2019-11-09:
From just looking at that it looks like the version of `click` is failing to throw an exception. We'll fix that.

But regardless even with a fix, I think you'd still get a different error if you ran what is in your script. You are using `--to dss` when I think you should be using `--to opendss`.

spark706, 2019-11-09:
With the exception failure issue aside I did run the code per your comment of using `--to opendss`

`(py2) C:\Users\shepa\DITTO\ditto-master\examples>ditto-cli convert --from cyme --input ieee_123node_cyme_to_opendss.json --to opendss --output ./
No handlers could be found for logger "ditto.writers.abstract_writer"
c:\users\shepa\anaconda3\envs\py2\lib\site-packages\ditto\models\base.py:32: UserWarning: Duplicate name 450-451 being set. Object overwritten.
  warnings.warn("Duplicate name %s being set. Object overwritten." % name)`

Please note that for this problem (the the second I have posted also) I did nothing more that create a Python 2.7 environment in Anaconda and then changed the directory to the folder with the downloaded scripts and ran them, including copying and pasting the example  code or file names and not typing them at all.

May I recommend that if there is a difference between ` --to dss` and ` --to opendss` that be noted in the documentation?

PR #313 (2020-04-28): Change click.BadOptionUsage to take two arguments
Fixes #312
closed 2020-04-28 by kdheepak