NatLabRockies/ditto #314: Error while converting IEEE4 testcase from CYME to OpenDSS

thakars, 2019-12-06:
Hi All,

I was trying to convert the IEEE4 testcase from tests/data/small_cases/ieee_4node/

I am getting the following error while parsing the model after reading the cyme files. For conversion, I am using 'ieee_123node_cyme_to_opendss.py' from examples folder with the input and output path suitably modified.

I am getting the following error:

Warning. The current CYME--->DiTTo reader was developed with documentation of CYME 8.0. Your version is 8.00. You might want to check the differences between the two.
Traceback (most recent call last):
  File ".\ieee_123node_cyme_to_opendss.py", line 59, in <module>
    main()
  File ".\ieee_123node_cyme_to_opendss.py", line 40, in main
    r.parse(model)
  File "C:\Users\Normal Account\AppData\Local\Programs\Python\Python37\lib\site-packages\ditto\readers\cyme\read.py", line 904, in parse
    modifier = system_structure_modifier(model)
  File "C:\Users\Normal Account\AppData\Local\Programs\Python\Python37\lib\site-packages\ditto\modify\system_structure.py", line 97, in __init__
    self.G.set_attributes(self.model)
  File "C:\Users\Normal Account\AppData\Local\Programs\Python\Python37\lib\site-packages\ditto\network\network.py", line 254, in set_attributes
    self.graph.node[i.name][attr] = getattr(i, attr)
AttributeError: 'Graph' object has no attribute 'node'

tarekelgindy, 2019-12-11:
Hi @thakars  and apologies for the slow reply.

It looks like we were using the deprecated node method from networkx.Graph which has now been deprecated in networkx 2.4: https://networkx.github.io/documentation/stable/release/release_2.4.html#deprecations

The quick fix is to revert to an older version of networkx using a command like this:
` pip install 'networkx==2.2' `

We'll work on making the actual fix of replacing the "node" functions with "nodes" as advised by the announcement. I'll close this issue once that fix has been merged into Master.

thakars, 2019-12-11:
Hi @tarekelgindy 

No worries about the slow reply. Using networkx 2.2 temporarily fixed the issue for me, thank you!