NatLabRockies/ditto #315: CYME to OpenDSS

chumontano, 2019-12-06:
Hi!

I am trying to use DiTTo to translate feeder data from CYMEDIST to OpenDSS. So, first I tried to run some examples to understand the data I am using as input and what could obtain as output. For that reason, I am using the code called:

"test_cyme_to_opendss.py" that is located in \ditto-master\tests

But, when I try to run this file for the IEEE4, I am getting the following error:

PS \ditto-master> python .\test_cyme_to_opendss.py
cyme_to_opendss.py
>>> Reading from CYME...
Unable to open file \ditto-master\cyme_to_opendss.py\network.txt
Traceback (most recent call last):
  File ".\test_cyme_to_opendss.py", line 62, in <module>
    test_cyme_to_opendss()
  File ".\test_cyme_to_opendss.py", line 38, in test_cyme_to_opendss
    r.parse(m)
  File "\ditto-master\ditto\readers\cyme\read.py", line 884, in parse
    self.parse_header()
  File "\ditto-master\ditto\readers\cyme\read.py", line 953, in parse_header
    "Could not find [SI] or [IMPERIAL] unit system information. Unable to parse."
ValueError: Could not find [SI] or [IMPERIAL] unit system information. Unable to parse.

I have been trying different approaches but I have not been able to get the translation, not even for the most simple case. 

Is there any suggestion, or another way you know to translate data from cyme to opendss format?

Thanks,

etimberg, 2019-12-07:
@chumontano are you able to share your cyme files? That error occurs because it's unclear what the units in the cyme file are.

tarekelgindy, 2019-12-11:
Hi @chumontano and sorry for the slow reply

So our CYME error handling for missing files isn't ideal at the moment. The first thing that our CYME reader looks for is the units, and hence this error often appears when the file isn't there.

From your message it looks like you took the file test_cyme_to_opendss.py and moved it to the base folder you're running. The file test_cyme_to_opendss.py configured the reader with the location of the input data. This is written as:

`r = Reader(data_folder_path=os.path.join(current_directory, 'data/small_cases/cyme',model))`

which for the 4-node feeder would be:

`r = Reader(data_folder_path=os.path.join(current_directory, 'data/small_cases/cyme/ieee_4node'))`

from the `tests` folder. If you're running this from your base folder you'll need to create it with:

`r = Reader(data_folder_path=os.path.join(current_directory, 'tests/data/small_cases/cyme/ieee_4node'))`

Alternatively you could move the files to the location where you're conversion script is, since the default location to look for is the current directory.

Does this fix your problem?

chumontano, 2019-12-11:
Hi, my CYME files are located in the same folder as the script I am running, also, I took care of take the path where they are and assigned to the reader. I am still having problems using Ditto, as I read in another discussion, it could be because I am using a most recent version of CYME (8.2). Thank you.

scmerrim, 2019-12-18:
Could be related to changes to the _CYMEQSOURCE_ table. 

In CYME 8.2 they added the capability to handle two different source impedances.  Instead of _ZEROSEQUENCEREACTANCE_ for example, they are now using two fields: _FIRSTLEVELX0_, and _SECONDLEVELX0_.  

New columns are now:
- FIRSTLEVELX0
- FIRSTLEVELX1
- FIRSTLEVELX2
- FIRSTLEVELR0
- FIRSTLEVELR1
- FIRSTLEVELR2
- SECONDLEVELX0
- SECONDLEVELX1
- SECONDLEVELX2
- SECONDLEVELR0
- SECONDLEVELR1
- SECONDLEVELR2