NatLabRockies/ditto #316: Test files for conversion from Synergi Electric to OpenDSS

emesk, 2020-01-22:
Hello,

I am trying to convert a Synergy network to OpenDSS. The conversion is successful but the load flow results in voltages way out of ranges. 

Can someone share a test network in Synergi format (network + warehouse files or integrated MS access database) that I can convert using Ditto and compare it with my Synergi network . 

Thanks

Best regards
Sohail

wenbowangnrel, 2020-01-24:
Hi Emest,

First, try your load flow in Synergi, if it can be run, meaning your network+warehouse is correct.
Second, Check load.dss in the converted folder, is nominal voltage defined?

emesk, 2020-01-27:
Hi @wenbowangnrel Thanks for the reply. I have load flow results from synergi and consist of line loading and losses, which are in acceptable ranges. However, I don't have yet voltage information. Considering your suggestions, I have checked load.dss in the converted file, the voltage levels are not defined there. I also checked synergi read and write scripts. I found that the load objects do not have nominal_voltage value. I traced it to the network.mdb file, and could not find voltage information for loads there. If someone can share sample synergi files it will be easier to know what essential information should be there in network.mdb and warehouse.mdb files.

wenbowangnrel, 2020-01-27:
Hi Emesk,

This not defining nominal kv for load.dss comes from the fact that synergi defines nominal voltage for lines instead of loads. You can find that in InstSection in network.mdb. For you to finish your task, you can try to colon my branch on Github and re-convert where I have fixed the issue, but did not merge to the main branch yet.

emesk, 2020-01-28:
Hi @wenbowangnrel I have tried your branch, and found that it read the voltage level from InstSection Description, which is hard-coded. As the case with my network file, the desciption does not include voltage information. I don't know from where else I can map the voltage levels. I can share my network if you can help me please, if so please write me short email, i will respond with my files there.

wenbowangnrel, 2020-01-29:
Yes if you want, you can sent me a sample of your network at Wenbo.Wang@nrel.gov

wenbowangnrel, 2020-01-31:
Hi Sohail,

If you only want to finish your task, in Ditto, there is a function to calculate nominal_voltage, you can call it in you script, like this:
   
```
from ditto.modify.system_structure import system_structure_modifier
modifier = system_structure_modifier(self.model, source)   
modifier.set_nominal_voltages()
```
This link of the function is here: https://github.com/NREL/ditto/blob/master/ditto/modify/system_structure.py#L524

Also, another valid question is how Synergi deals with nominal_voltage? I am digging into this right now, will share more information later.

I want to thank you for bringing this up.

Best,
Wenbo

emesk, 2020-02-05:
Hi Wenbo,

Thanks for inputs. I have used this conversion in the script in converter.py in between read and write calls

`        
        self.reader.parse(self.m)

        if self.jsonize:
            self.json_writer = self.json_writer_class(output_path=self.json_path)
            self.json_writer.write(self.m)

        from ditto.modify.system_structure import system_structure_modifier
        modifier = system_structure_modifier(self.m)
        modifier.set_nominal_voltages()

        self.writer.write(self.m)`

However, i have errors in network.py as graph does not have node attribute

`                            self.graph.node[i.name][attr] = getattr(i, attr)
                            self.digraph.node[i.name][attr] = getattr(i, attr)`

Following error message is encountered. I dont know where exactly should i call the functions you have mentioned. 

`    self.graph.node[i.name][attr] = getattr(i, attr)
AttributeError: 'Graph' object has no attribute 'node'
`

Thanks alot for your time
Sohail

wenbowangnrel, 2020-02-09:
Hi Sohail,

Got it, please allow us some time to take a look at your error.

Best,
Wenbo

tarekelgindy, 2020-02-10:
Hi Sohail,

This is a known bug that we need to fix. The temporary workaround is addressed in this issue:
https://github.com/NREL/ditto/issues/314

To quote the error:

It looks like we were using the deprecated node method from networkx.Graph which has now been deprecated in networkx 2.4: https://networkx.github.io/documentation/stable/release/release_2.4.html#deprecations

The quick fix is to revert to an older version of networkx using a command like this:
` pip install 'networkx==2.2' `

We'll work on making the actual fix of replacing the "node" functions with "nodes" as advised by the announcement. I'll close this issue once that fix has been merged into Master.

I'll push this up on my priority list and update you when we've made the fix.

emesk, 2020-02-12:
Thanks for the hint @tarekelgindy I have done the changes, I am still finding difficulty. Maybe the Synergee network is missing something important for the conversion process. I shall wait for @wenbowangnrel feedback. Thats why it could be useful if we have some test files for Synergee that we can compare with our network files to find missing data.

wenbowangnrel, 2020-02-12:
Hi @emesk, 

It is not easy to have test files for Synergi shared, there are some legal processes. But I think as long as you can run it in Synergi, your network should be okay. 

As far as the question that how Synergi calculates nominal voltage for the line: Synergi uses the section based method to calculate power flow: the so called backward forward sweep for radial systems. So the nominal voltage is initially defined in the InstFeeders tab in the network.mdb, and then as the iteration process goes on, the nominal voltage changes whenever the iteration encounters a transformer. 

However, Opendss uses current injection power flow formulation and fix-point iteration to solve power flow, .

In my Network model, there is nominal voltage defined in the InstSection. For you to get the nominal voltage, you can create a function to calculate the nominal voltage (similar to Synergi or DiTTo graph function), or you can manually put them in the InstSection tab.

Hope it helps,
Wenbo

jeffsman, 2024-03-16:
Hello Wenbo and everyone in this forum. 

I am currently doing my research using OpenDSS. I will be using .mdb files from Synergi. Is there a direct way to load such files in OpenDSS now?

Looking forward to your prompt response.

Best regards,

Jeff

wenbowangnrel, 2024-03-28:
No such direct way to load mdb in opendss as far as I know.

KayyagariBMcD, 2025-03-07:
Hello Wenbo,
I currently use different versions of CYME (CYME 9.2 to CYME 9.4) and am particularly interested in converting Synergi models to CYME.

I also noticed that DiTTo supports CYME 8.0. Could you provide more insight into its capabilities for this specific conversion? Specifically, does DiTTo support direct Synergi-to-CYME conversion, or would an intermediate step (e.g., converting to OpenDSS first) be required? Also, do you know if there are any limitations or considerations we should be aware of when using DiTTo for this purpose?

Thanks,