NatLabRockies/ditto #321: `AttributeError: 'NoneType' object has no attribute 'lower'`

kdheepak, 2020-04-30:
```
Traceback (most recent call last):
  File "13node_ephasorsim.py", line 33, in <module>
    w.write(m)
  File "/Users/USER/gitrepos/ditto/ditto/writers/ephasor/write.py", line 1369, in write
    df4 = self.transformer()
  File "/Users/USER/gitrepos/ditto/ditto/writers/ephasor/write.py", line 779, in transformer
    i.from_element + "_" + pw.phase.lower()
AttributeError: 'NoneType' object has no attribute 'lower'
```

kdheepak, 2020-04-30:
It seems like when opendss reader parses a file, it doesn't set the phase always. Is this a bug?

tarekelgindy, 2020-05-01:
Yes this is a bug. It relates to the representation of grounding in transformers. A transformer with bus represented with `bus=XXX.1.2.3.0` is a three-phase grounded transformer, but the OpenDSS reader is interpreting .0 as a phase.

The correct action here is to add an attribute to the Winding class titled `is_grounded` or something similar. When parsing  OpenDSS transformers, the reader should look for .0 and set the flag to be true if need be. This was a future to-do idem indicated by #301