NatLabRockies/ditto #330: GridLAB-d to OpenDSS conversion

asavasci, 2020-06-13:
I am trying to convert IEEE_4_node gridlab-d model into an opendss format using ditto-cli tool. However, I am getting the following error
`Log file currently not supported, please contact the developers for information on how to generate log files`

I would be glad if you guide me on this.

kdheepak, 2020-06-13:
Can you paste here what you are trying to run? If you can share a
reproducible example that would be great.

asavasci, 2020-06-13:
This is the file that I am trying to convert
[IEEE_4_node.zip](https://github.com/NREL/ditto/files/4774019/IEEE_4_node.zip)
ditto-cli converts the glm file attached here, however solution does matches with original solution.
[OpenDSS_modelbyditto.zip](https://github.com/NREL/ditto/files/4774021/OpenDSS_modelbyditto.zip)

asavasci, 2020-06-13:
This is the file that I am trying to convert
IEEE_4_node.zip
ditto-cli converts the glm file attached here, however solution does matches with original solution.
OpenDSS_modelbyditto.zip

kdheepak, 2020-06-13:
How are you trying to convert it? Are you running it from the command line? Can you share the full stacktrace or a screenshot?

asavasci, 2020-06-13:
Yes, using command line
`ditto-cli convert --from gridlabd --input C:\Users\alper\Desktop\grid4node\IEEE_4_node.glm --to opendss --output C:\Users\alper\Desktop\grid4node`
and getting 
`Log file currently not supported, please contact the developers for information on how to generate log files`

asavasci, 2020-06-13:
I modified generated OpenDSS files, then I obtained meaningful but not same results. The issues that I am seeing in generated files are `sourcebus` cannot be connected to the `node1` automatically. Thus, I modified the line `New Line.node1-node2 Units=km Length=0.6096 bus1=sourcebus.1.2.3 bus2=node2.1.2.3 switch=n enabled=y phases=3 Linecode=3P_OH_Code0`  specifying the sourcebus as the sending end bus of the first line segment.   

Also modifying, ``New Circuit.Full_Network`  as `New Circuit.Full_Network basekV=12.47 phases=3`.

[OpenDSS_2_byditto.zip](https://github.com/NREL/ditto/files/4774142/OpenDSS_2_byditto.zip)