NatLabRockies/ditto #334: "Vsource.source" KeyError when instantiating NetworkAnalyzer on OpenDSS model

petergoldthorpe, 2020-06-23:
Hi, I'm a colleague of @frederikgeth. We're trying to use DiTTo for a project that uses the metrics command with an opendss file, and we are running into the following error. We've narrowed down the problem to an issue with instantiating the `NetworkAnalyzer` model, and have been table to replicate for a test case that comes packaged with DiTTo. The test case is located at `./tests/readers/opendss/Powersource/test_powersource.dss`

My code is as follows:
```python
from ditto.store import Store
from ditto.readers.opendss.read import Reader
from ditto.metrics.network_analysis import NetworkAnalyzer

model = Store()

filepath = 'C:/Users/GOL135/projects/ditto-public/ditto/tests/readers/opendss/Powersource/'
opendss_reader = Reader(master_file=filepath+'test_powersource.dss', buscoordinates_file=filepath+'buscoord.dss')
opendss_reader.parse(model)
# model.set_names()

network_analyst = NetworkAnalyzer(model=model) #<--- Error
network_analyst.compute_all_metrics()
```
Which throws the following error
```python
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm 2020.1.2\plugins\python\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm 2020.1.2\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:/Users/GOL135/AppData/Roaming/JetBrains/PyCharm2020.1/scratches/scratch.py", line 40, in <module>
    network_analyst = NetworkAnalyzer(model=model)
  File "C:\Users\GOL135\projects\ditto-public\ditto\ditto\metrics\network_analysis.py", line 160, in __init__
    modifier.set_nominal_voltages()
  File "C:\Users\GOL135\projects\ditto-public\ditto\ditto\modify\system_structure.py", line 583, in set_nominal_voltages
    upstream_transformer_name = self.G.get_upstream_transformer(
  File "C:\Users\GOL135\projects\ditto-public\ditto\ditto\network\network.py", line 357, in get_upstream_transformer
    edge_type = edge_equipment[(curr[0], curr_node)]
KeyError: ('Vsource.source', 'sourcebus')
```
The same error is thrown when using the `model.set_names()` method. We have used the traceback to find the error in the `./ditto/network/network.py` file.
```python
def get_upstream_transformer(self, model, node):

    curr_node = node
    curr = list(self.digraph.predecessors(node))
    edge_equipment = nx.get_edge_attributes(self.digraph, "equipment")
    edge_equipment_name = nx.get_edge_attributes(self.digraph, "equipment_name")
    # import pdb; pdb.set_trace()
    while curr != []:
        edge_type = edge_equipment[(curr[0], curr_node)]
        if edge_type == "PowerTransformer":
            return edge_equipment_name[(curr[0], curr_node)]
        curr_node = curr[0]  # assuming that the network is a DAG
        curr = list(self.digraph.predecessors(curr_node))
    return None
```
---
Here the code throws a `KeyError`. The `edge_equipment` is a dictionary of tuples, and the code fails when it attempts to call a tuple which contains `'Vsource.source'` as the first element. No such element exists in the `edge_equipment` dictionary.

In the `./tests/readers/opendss/Powersource/test_powersource.py` there is a function which reads in an OpenDSS file and asserts that
```python
model["Vsource.source"].name == "Vsource.source"
```

This assertion passes with no error on my machine, but I still encounter the `KeyError` when using my own code as when I print `model.name` I encounter:
```python
AttributeError: 'Store' object has no attribute 'name'
```

Any advice would be welcome.
Kind regards, Peter

kdheepak, 2020-06-23:
Thanks for the detailed issue.

I think in your code where you use `model.name`, you are instead doing `store.name`, and that what is giving this error. Can you try `print(store.elements)`? That should return a list of "models".

petergoldthorpe, 2020-06-23:
Sure, I've tried printing the commands from my code. The output is as follows:
```python
>>> print(model)
<ditto.store.Store(elements=0, models=28) object at 0x21185c22a30>
>>> print(model.elements)
[]
>>> print(model.models)
(<ditto.models.power_source.PowerSource object at 0x0000024FBFED3CD0>, ... ) #tuple
```

These print statements are thrown before the KeyError.

thaitranp, 2020-09-05:
I'm also running into the same key error issue when trying to call build_networkx() for the ephasor conversion test.

jnh277, 2022-02-01:
same issue when trying to use build_networkx()