NatLabRockies/ditto #335: 'Cannot write to format dss'

twigandrud, 2020-07-20:
I'm trying to use DiTTo to convert from a Cyme database to OpenDSS format. I believe I have the command syntax correct, but when I run the command line command to run DiTTo, I get the error "Cannot write to format 'dss'"

Here's the command I put into the Windows command line:
>ditto-cli convert --input C:\test\config.json --from cyme --to dss --output C:\test

I've attached the JSON file I created because I hadn't created one before, so it could possibly in a format that won't work with DiTTo. If that's the case, I'd appreciate any assistance formatting it correctly. I added the ".txt" so that I could upload it to Github.
The files referenced in the JSON file were created using CYME 9.0. I'm running Windows 10 64-bit. Thanks.

[config.json.txt](https://github.com/NREL/ditto/files/4949911/config.json.txt)

kdheepak, 2020-07-20:
Your json file just contains:

```
{
	"data_folder_path": "C:\test",
	"network_filename": "net.txt",
	"equipment_filename": "equp.txt",
	"load_filename": "loads.txt"
}
```

Can you share the full stacktrace? It'll help identify at what step there was an error.

twigandrud, 2020-07-21:
Is the json file supposed to contain more or something else? I tried following some steps online and that's what I got.

The only message I get in the command window is:
>Error: cannot write to format 'dss'

If DiTTo outputs errors to a text file, I'd be happy to provide it, but otherwise that's the only message I get.

kdheepak, 2020-07-21:
Interesting. I'm pinging @tarekelgindy to see if he has any ideas. I'll take a closer look when I get the chance this week.