NatLabRockies/ditto #368: codecov bash uploader security update

NicolasGensollen, 2021-04-16:
Hi Ditto team, :wave: 

You probably already know that, but Codecov.io announced yesterday that their bash uploader script had been compromised: https://codecov.io/disclosure

AFAICT, this script is used by the codecov action which means that authentication secrets used in the same workflow (or potentially other workflows) may have been compromised.

https://github.com/NREL/ditto/blob/a1a527fe043258aacfab8ab84ff162f84a8272da/.github/workflows/ci.yml#L33-L37

There is ongoing work to include a checksum validation of the script within the codecov action, but it might be worth having a look!

tarekelgindy, 2021-04-20:
Thanks Nicolas! I believe that Dheepak has updated the secret tokens.
Great to hear from you!!

NicolasGensollen, 2021-04-21:
Hi @tarekelgindy 

 >I believe that Dheepak has updated the secret tokens.

Great, I'll close this then! :+1:  
Note that codecov merged a checksum validation fix very quickly. So this action should hopefully be safe now. 

>Great to hear from you!!

Thanks! Great to see that you're still maintaining this package! 
Hope everything is fine for you!
Cheers! :smiley:

closed 2021-04-21 by NicolasGensollen

kdheepak, 2021-04-21:
Thanks for bringing this up. I didn't get any email about this so I didn't think we were affected. But I've regenerated tokens for all the secrets we have just in case.