NatLabRockies/ditto #374: Conversion from CYME 9.1 to OpenDSS

hputra728, 2021-09-13:
Hi I'm doing a conversion from CYME 9.1.0 to OpenDSS. I encountered a format problem when I tried to find the conversion results in OpenDSS. I did the conversion manually but when I tried to run it again in OpenDSS I got mixed results with the simulation on CYME 9.1.0 . Is there a specific version converting from CYME 9.1.0 to OpenDSS ?

tarekelgindy, 2021-10-20:
Apologies for the slow response. What are the specific issues that you're having? We integrated changes that should support conversion from CYME 9.1.0. If this is still a problem, we can address the components that seem to be problematic.

albegehtv, 2022-04-13:
no reconoce los tranformadores y las cargas .. sale error parse