NatLabRockies/ditto #376: conversion from cyme to opendss ( command line)

mwague, 2021-09-18:
I am having this error while converting cyme 9.1 to opendss on my command line
TypeError: `trait` must be a Trait or None, got (<ditto.models.base.Int object at 0x000001D5D69C8070>, <ditto.models.base.Int object at 0x000001D5D69C80D0>, <ditto.models.base.Float object at 0x000001D5D69C8100>) <class 'tuple'>

What maybe the potential issues please?

kdheepak, 2021-09-18:
Can you please share the full stack trace?

mwague, 2021-09-19:
[
[image]
](url)

mwague, 2021-09-19:
Here is the picture

mwague, 2021-09-26:
any follow up?

tarekelgindy, 2021-10-20:
Sorry for the slow responsiveness with this.
The most likely cause of this is that we were using a few deprecated methods in the traitlets library that  outdated in their latest release. However, these issues should have been addressed by [this](https://github.com/NREL/ditto/pull/358) pull request.

Are you using the most recent version of ditto? If not, try running
`pip install ditto.py==0.2.2`

which will update to the latest ditto release.