NatLabRockies/ditto #377: cyme to opendss ( command line) or ditto issue

mwague, 2021-09-28:
I am having this error while converting cyme 9.1 to opendss on my command line

**TypeError: trait must be a Trait or None, got (<ditto.models.base.Int object at 0x000001D5D69C8070>, <ditto.models.base.Int object at 0x000001D5D69C80D0>, <ditto.models.base.Float object at 0x000001D5D69C8100>) <class 'tuple'>**

What maybe the potential issues please?

mwague, 2021-09-28:
[image]

albegehtv, 2022-01-11:
hola como que tal como esta tu archivo conv.py configurado?? por el metodo 2???

albegehtv, 2022-01-11:
Yo estaba queriendo ayuda tambien para convertir los archivos de Cyme a Opendss, pero no consigo de ninguna forma.