NatLabRockies/ditto #394: File conversion from

saheb26-dev, 2022-06-13:
Hey, 
I'm trying to convert a power grid model I created in OpenDSS to Gridlab-D. I'm using Ditto for the conversion and getting the following error. Running root so not very sure how access can be an issue. Getting the same error when not running root.
OS: Rocky Linux

[image]

PMeira, 2022-07-19:
I couldn't reproduce that. Is `dssfile` an actual file? Or a folder?

saheb26-dev, 2022-08-01:
It's the folder which contains dss files.

PMeira, 2022-08-01:
@saheb26-dev You need to pass the main/master .dss script path, not the folder. Basically you instructed the program to try to open the folder as a file, hence the error. Note from the error message that it expects a file, no mention of folder.

On why you need a file instead of a folder: OpenDSS has scripting capabilities so a single folder can contain multiple scenarios and studies. For converting, you typically just need the data commands that setup the base circuit.

saheb26-dev, 2022-08-01:
@PMeira Thank you so much for your input. Worked once I made the changes.

closed 2022-08-01 by saheb26-dev