NatLabRockies/ditto #395: Conversion issue form OpenDSS-G/OpenDSS to Gridlab-D

Barsha96, 2022-07-15:
Hello, I am trying to convert openDSS files (Using Master.dss) to the Gridlab-d glm file. I am using the following code: 

`ditto-cli convert --from opendss --input Master.dss --to gridlabd --output ./out`

I can convert a IEEE_4_node.glm into opendss files but cannot convert the same dss files into a glm file. The Issues I get are: 

```
Traceback (most recent call last):
  File "c:\python3.9\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python3.9\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Python3.9\Scripts\ditto-cli.exe\__main__.py", line 7, in <module>
  File "C:\Users\bupadhy\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\bupadhy\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\bupadhy\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\bupadhy\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\bupadhy\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\bupadhy\AppData\Roaming\Python\Python39\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python3.9\lib\site-packages\ditto\cli.py", line 173, in convert
    Converter(
  File "c:\python3.9\lib\site-packages\ditto\converter.py", line 246, in convert
    self.reader.parse(self.m)
  File "c:\python3.9\lib\site-packages\ditto\readers\opendss\read.py", line 245, in parse
    self.parse_power_source(
  File "c:\python3.9\lib\site-packages\ditto\readers\opendss\read.py", line 52, in timed
    result = method(*args, **kw)
  File "c:\python3.9\lib\site-packages\ditto\readers\opendss\read.py", line 344, in parse_power_source
    sources = _dss_class_to_dict("Vsource")
  File "c:\python3.9\lib\site-packages\ditto\readers\opendss\read.py", line 2765, in _dss_class_to_dict
    return dss.utils.class_to_dataframe(class_name).to_dict(orient="index")
  File "c:\python3.9\lib\site-packages\opendssdirect\utils.py", line 125, in class_to_dataframe
    dss.Circuit.SetActiveClass("{class_name}".format(class_name=class_name))
  File "c:\python3.9\lib\site-packages\opendssdirect\Circuit.py", line 100, in SetActiveClass
    return CheckForError(lib.Circuit_SetActiveClass(ClassName))
  File "c:\python3.9\lib\site-packages\dss\_cffi_api_util.py", line 81, in CheckForError
    raise DSSException(error_num, self._get_string(self._lib.Error_Get_Description()))
dss._cffi_api_util.DSSException: (242, 'Redirect File: "C:\\Users\\bupadhy\\OneDrive.dss" Not Found.')
```
The system says OneDrive.dss is not found. The system seems to be automatically getting to that file that was not provided and that does not exist.

Is there anything that I might be doing wrong? 

Thanks and Regards, 
Barsha

PMeira, 2022-07-19:
@Barsha96 Any chance you have spaces in your full file paths?

There's probably a pair of quotes missing in DiTTo.

kdheepak, 2022-07-19:
@PMeira where do you think a pair of quotes is missing?

PMeira, 2022-07-19:
@kdheepak Right here:

https://github.com/NREL/ditto/blob/a1dcd4354bdc9dd25e92216498c4c3d17b4cb22e/ditto/readers/opendss/read.py#L201

Barsha96, 2022-07-21:
> @Barsha96 Any chance you have spaces in your full file paths?
> 
> There's probably a pair of quotes missing in DiTTo.

The issue was I had to give full path and quote those with spaces and dashes specially for the input. I had to give absolute path for the input but could give both absolute path and relative path fir the output location.

`ditto-cli convert --from opendss --input "C:\Users\bupadhy\OneDrive - UTol\Desktop\conversion\CampusModel\ModelDSS\Master.dss" --to gridlabd --output ./CampusModel/converted`

kdheepak, 2022-07-21:
@PMeira I can make a PR for that. Thanks for pointing that out!