NatLabRockies/ditto #396: Could not set nominal voltage for bus sourcebus - Opendss to glm conversion

Barsha96, 2022-07-21:
Hi, I am using Ditto to convert from OpenDSS-G to Gridlab-D. I am facing a lot of issues doing that. I am not sure if there is any information I am missing. 

Details: 
When I convert OpenDSS-G's master file to glm using the following command: 
`ditto-cli convert --from opendss --input "C:\Users\bupadhy\OneDrive - UTol\Desktop\conversion\CampusModel\ModelDSS\Master.dss" --to gridlabd --output ./CampusModel/converted `

I get two errors: 
```
Log file currently not supported, please contact the developers for information on how to generate log files
Could not set nominal voltage for bus sourcebus
```
But I can see a converted glm files. However, for whichever OpenDss model I convert, It gives out the same Model.glm. 

Model.glm:
```
module powerflow{
    solver_method NR;
    NR_iteration_limit 50;
};

object node {
    name nnode1;
    phases ABCN;
     nominal_voltage 2400.0;
};

object node {
    name nnode2;
    phases ABCN;
     nominal_voltage 2400.0;
};

object node {
    name nnode3;
    phases ABCN;
     nominal_voltage 2400.0;
};

object node {
    name nload4;
    phases ABCN;
     nominal_voltage 2400.0;
};

object load {
    name nload_load_load4;
    nominal_voltage 2400.0;
    parent nload4;
    constant_power_A 1800000+871779.7887081344j;
    constant_power_B 1800000+871779.7887081344j;
    constant_power_C 1800000+871779.7887081344j;
};

object transformer_configuration {
    no_load_loss 0.0;
    connect_type WYE_WYE;
    primary_voltage 12470.0;
    secondary_voltage 4160.0;
    power_rating 2000.0;
    reactance 0.06;
    resistance 0.01;
    name transformer_config_1;
};

object transformer{
    name ntransformer23;
    from nnode2;
    to nnode3;
    phases ABC;
    configuration transformer_config_1;
};

object line_configuration {
    z11 0.00031054140000000004+0.0005860069j;
    z12 0.00012386719999999999+0.0003157983j;
    z13 0.0001219249+0.0002441818j;
    z21 0.00012386719999999999+0.0003157983j;
    z22 0.0003176462+0.0005641492j;
    z23 0.0001254668+0.0002665902j;
    z31 0.0001219249+0.0002441818j;
    z32 0.0001254668+0.0002665902j;
    z33 0.0003136219+0.0005764944j;
    name line_config_1;
};

object overhead_line{
    length 2000.000064;
    configuration line_config_1;
    name nnode1-node2;
    from nnode1;
    to nnode2;
    phases ABC;
};

object overhead_line{
    length 2500.00008;
    configuration line_config_1;
    name noverhead_line:34;
    from nnode3;
    to nload4;
    phases ABC;
};
```
Are there any preliminaries to be understood before converting OpenDSS-G's Master file to a Gridlab-d Model? I have attached the OpenDSS-G model I am working with and the glm file converted by ditto. I would appreciate any help in this.

[converted.zip
[ModelDSS.zip](https://github.com/NREL/ditto/files/9162699/ModelDSS.zip)
](https://github.com/NREL/ditto/files/9162698/converted.zip)

PMeira, 2022-07-22:
@Barsha96 The .dss files are broken:

- This doesn't work (`.3_1`): `New "Line.sw33" phases=3 bus1=b31.1.2.3 bus2=b31n1.1.2.3_1 Switch=True`
- Capacitors are using `kV=0`:
```
New "Capacitor.c1" phases=3 conn=wye kv=0 bus1=b7n2.1.2.3 bus2=b7n4.1.2.3
New "Capacitor.cap_1" phases=3 conn=wye kv=0 Numsteps=3 bus1=b7n6.1.2.3 bus2=b7n7.1.2.3
```

Still, I tested DiTTo with OpenDSSDirect.py v0.6.1 and the latest v0.7 and both reported errors.
Before OpenDSSDirect.py v0.7, `GIScoords` command should also fail, per #387.

I recommend trying to run in official OpenDSS directly since it will also report various errors there.

(@kdheepak Hope you don't mind me checking this, just making sure our everything is fine on ODD.py)

Barsha96, 2022-07-22:
Thank you for your direction. I'll try and look for the errors in the dss model. 
Something I wanted to figure out, I tried converting IEEE_4_node.glm file to the dss format as I wanted to try with a working dss file, converted it back again to glm format. The conversion gave me the same file as the other models did and with the same error. I am searching for a good example dss files as well which will convert correctly to the glm. Anywhere I can get those? 

So far, I only have the dss models that are giving me the same exact file with the same content and the same error. If I get some other working dss file to convert, I might get more answers as well.
[IEEE_4_node.zip](https://github.com/NREL/ditto/files/9172415/IEEE_4_node.zip)