NatLabRockies/ditto #397: cannot handle Cyme v9 exports

nlaws-camus, 2022-08-23:
I recently used DiTTo to convert a v9 Cyme export to dss and see few issues in the dss files:
1. all loads have `kV=0`
2. no `new circuit` command
3. loads with `Phases=12`?
4. source `pu=None`
5. 
All of these issues could be related to how Cyme created the txt files. Before I go down the path of figuring out what needs fixed in DiTTo vs. what's wrong with the Cyme export I want to check with the DiTTo developers:

Is there a branch/fork under development at NREL for handling Cyme v9?

cc: @daniel-thom @kdheepak @tarekelgindy @

nlaws-camus, 2022-08-23:
If someone could point me to the documentation for the Cyme export format I could possibly contribute the changes needed for Cyme v9. I have looked through some of the Cyme user documentation and have not found any references yet. 

One possible difference for Cyme v8 vs. v9 I have found:
 - I see [`desiredvoltage` in the Cyme reader for a [SOURCE]](https://github.com/NREL/ditto/blob/a1dcd4354bdc9dd25e92216498c4c3d17b4cb22e/ditto/readers/cyme/read.py#L1044) 
 - but in the Cyme v9 export there is no `desiredvoltage`:
```
[SOURCE]
FORMAT_SOURCE=SourceID,DeviceNumber,NodeID,NetworkID,OperatingVoltageA,OperatingVoltageB,OperatingVoltageC,UseSecondLevelImpedance,SinglePhaseCenterTap,CenterTapPhase
``` 
Is this the kind of thing that will need updated for the new Cyme version? (eg. change `desiredvoltage` to `OperatingVoltageA`)

yuanyx23, 2022-12-16:
Hi, I recently use DiTTo to convert a V9 CYME export to dss and see a new issue. The error message shows "ValueError: No PowerSource object found in the model". I believe this is also caused by the new txt format of CYME. I would like to know how you solve your problem and see if I can follow the similar idea. Thanks!