NatLabRockies/ditto #421: Redirect file not found: "C:/Users/m.khan2/OneDrive"

StormsHalted, 2023-04-13:
Hello NREL,

I make a very simple code to convert opendss into Gridlab-d:

Note: I do not have a coordinates file for this model.

`from ditto.readers.opendss.read import Reader
from ditto.store import Store
from ditto.writers.gridlabd.write import Writer

model = Store()

dss_reader = Reader(master_file='C:/Users/m.khan2/OneDrive - Washington State University (email.wsu.edu)/UI-Assist/AvistaF7 - 2.dss'.format(model=model))

dss_reader.parse(model)`

I get this error:
`C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\python.exe C:\Users\m.khan2\PycharmProjects\DSStoGLD\DSStoGLD.py 
Traceback (most recent call last):
  File "C:\Users\m.khan2\PycharmProjects\DSStoGLD\DSStoGLD.py", line 9, in <module>
    dss_reader.parse(model)
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\ditto\readers\opendss\read.py", line 245, in parse
    self.parse_power_source(
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\ditto\readers\opendss\read.py", line 52, in timed
    result = method(*args, **kw)
             ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\ditto\readers\opendss\read.py", line 344, in parse_power_source
    sources = _dss_class_to_dict("Vsource")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\ditto\readers\opendss\read.py", line 2765, in _dss_class_to_dict
    return dss.utils.class_to_dataframe(class_name).to_dict(orient="index")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\opendssdirect\utils.py", line 123, in class_to_dataframe
    dss.Circuit.SetActiveClass("{class_name}".format(class_name=class_name))
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\opendssdirect\Circuit.py", line 98, in SetActiveClass
    return CheckForError(lib.Circuit_SetActiveClass(ClassName))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\m.khan2\AppData\Local\Programs\Python\Python311\Lib\site-packages\dss\_cffi_api_util.py", line 145, in _check_for_error
    raise DSSException(error_num, self._get_string(self._lib.Error_Get_Description()))
dss._cffi_api_util.DSSException: (#243) Redirect file not found: "C:/Users/m.khan2/OneDrive"`

My redirects are arranged as follows in the opendss model and all of them are in the same folder:

`Redirect Feeder.dss
Redirect Microgrid.dss
Redirect LoadShapes.dss
Redirect PV-Insertions.dss           ! **** File contains all PV - insertions as Generators
! Redirect PV-Insertions-PV.dss      ! **** File contains all PV - insertions as PVs
Redirect ProfileAssignement.dss`

PMeira, 2023-04-13:
It seems the fix for the issue with spaces in paths is not out yet, see https://github.com/NREL/ditto/issues/395#issuecomment-1189535398

> `C:/Users/m.khan2/OneDrive - Washington State University (email.wsu.edu)/UI-Assist/AvistaF7 - 2.dss`

Avoid spaces in your filenames and paths in general, and it should work fine (if it doesn't, try removing the parentheses too).

StormsHalted, 2023-04-14:
Hello PMeira,

I simplified the file path and I am now able to convert opendss to GLM successfully. 

However, Loads, Nodes, Transformers & Transformer configurations only are converted successfully but line conductor, line configuration, line spacing and lines did not convert into the converted glm file.

PMeira, 2023-04-14:
@StormsHalted Hm, I tested some files with the version from 2021 and it seems to fail like that too, but the latest version from GitHub worked for several circuits (some still failed but with other errors). 

So, if you're using the latest version on PyPI via pip, you could try installing the latest development version. First uninstall it (`pip uninstall ditto.py`), and then either get a local clone of the repo and `pip install -e .` from inside it, or just run the following to let pip download a snapshot:

```
pip install https://github.com/NREL/ditto/archive/refs/heads/master.zip
```

I imagine there could be some detail in your files that could still fail though.

StormsHalted, 2023-04-16:
Hello PMeira,

The latest development version is able to convert the model with line configuration and line conductors. Now, it appears that, about 70 - 80 % of my model is being converted without errors. In coming days, I will match it and let people know here if I find any conversion errors.

One, interesting thing I noted is that the line configuration block gets converted into sequence impedance as opposed to gmr and conductor resistance parameters.

Thanks for your help!