NatLabRockies/ditto #432: Traitlets version needs to be >=5.14 for Ladybug tools

vtnate, 2024-03-07:
[This dependency](https://github.com/NREL/ditto/blob/master/setup.py#L126) is causing problems for Ladybug Tools to use via Urbanopt.

Can it be bumped up to 5.14?

Alternatively, @tarekelgindy, was there discussion of removing Traitlets entirely from this library?

tarekelgindy, 2024-03-07:
Yes feel free.
We've been gutting ditto significantly and using an entirely new representation of the core. This is using two repos that are still on the private NREL github repo pending public release, and the developments to ditto are happening on a ditto fork on the intneral NREL github, so at some point there will be a major release that merges these changes in. We'll want to do testing of this with urbanopt before that happens obviously.

vtnate, 2024-03-08:
Great to hear this is getting some love!